KumarRobotics/msckf_vio #3: The Repo can work on Ubuntu 14.04 and ROS indigo?

YacongWang, 2018-01-09:
Thanks for your share. Can this  code  work on Ubuntu 14.04 and ROS indigo? Have you tested the indigo and Ubuntu 14.04?

kartikmohta, 2018-01-09:
I don't think this will work with Ubuntu 14.04 and ROS Indigo since we require some functions (related to the fisheye distortion model) from OpenCV which were only added in versions >=2.4.10 while Ubuntu 14.04/ROS Indigo only support OpenCV v2.4.8. Note that we've only tested this with Ubuntu 16.04 and ROS Kinetic/Lunar.

xushangnjlh, 2018-01-09:
I have tested on indigo and Ubuntu 14.04 and nodelet problem was found on image_processor topic.

w1019220367, 2018-01-09:
It can be run on ubuntu 14.04 and indigo. But you need to compile cv_bridge from source code with opencv3 and then set cv_bridge and opencv3 path into CMakelist.txt.

goldbattle, 2018-01-11:
I was able to run it on 14.04 (but at a lower performance compared to the videos) after I installed:
* [cv_bridge](https://github.com/ros-perception/vision_opencv) (indigo branch)
* [image_transport_plugins](https://github.com/ros-perception/image_transport_plugins) (indigo branch)
* `sudo apt-get install ros-indigo-opencv3`

And had some dynamic linking issues that I was able to fix (see https://github.com/KumarRobotics/msckf_vio/issues/5). I think I also had to make a symbolic link at one point between it wanting `/opencv-3.1.0/` and the default install location of just `/opencv/` in the include directories.

closed 2018-02-26 by YacongWang