KumarRobotics/msckf_vio #4: which topic could be showed on rviz?

YiChenCityU, 2018-01-09:


ke-sun, 2018-01-09:
The odometry topic should be `/$(arg robot)/vio/odom`.
Could you provide more details like which launch file and rviz configuration file you are using. Which dataset are you testing on? Is there any runtime error?

SkyRiderMike, 2018-01-10:
I don't know how to run this algorithm, could you please provide more details of command for running this code ?

YiChenCityU, 2018-01-10:
Hi,
I run like this:
roslaunch /home/eason/msckf_ws/src/msckf_vio/launch/msckf_vio_euroc.launch 
rosbag play /home/eason/Documents/data/MH_01_easy.bag

Rviz like this picture: [https://www.dropbox.com/s/5jx8i7na53ko4h1/Screenshot%20from%202018-01-10%2010%3A54%3A27.png?dl=0](url) 
[https://www.dropbox.com/s/epi9u5ixku1csos/Screenshot%20from%202018-01-10%2010%3A54%3A50.png?dl=0](url)

YacongWang, 2018-01-10:
@YiChenCityU Can you save the results about timestamp, position, and orientation? And the rviz don't plot the route just like VINS-mono. I think we have more to do to run this repo properly.

pxlong, 2018-01-10:
@YacongWang I agree. I have recorded a video on euroc dataset (V1_02_medium), the performance is not good. #6

cumtchenchang, 2018-01-10:
@ke-sun Can it output dataset trajectory like ORB-SLAM or vins-mono.

ke-sun, 2018-01-10:
@YiChenCityU 

There are some tips that may be helpful for you.

1. Make sure the repo is complied in release mode.
2. There are rviz configuration files in the rviz folder. It may be convenient to use them directly.
3. The filter requires the robot to stay static at first in order to collect a couple of IMU messages to initialize gyro bias, acc bias and most importantly, the initial orientation. For EuRoC `MH_01_easy` and `MH_02_easy`, the datasets start directly with the robot in motion. Therefore, the filter does not work well with those two datasets. Also for the other datasets, I suggest that you could launch the software first and then start to play the bagfiles, which allows the filter to catch the first few static IMU messages.

Hope that helps solve your problem.

ke-sun, 2018-01-10:
@cumtchenchang 

It does not output `nav_msgs/Path` as a topic. One possible quick way to visualize a "trajectory" is to accumulate the odometry messages in rviz.

closed 2018-01-15 by ke-sun

YacongWang, 2018-01-17:
I have added the ```nav_msgs/Path``` in the Publish function, and the ```echo topic``` is
```
---
header: 
  seq: 430
  stamp: 
    secs: 1403638540
    nsecs: 577829500
  frame_id: "world"
poses: 
  - 
    header: 
      seq: 0
      stamp: 
        secs: 1403638540
        nsecs: 577829500
      frame_id: "world"
    pose: 
      position: 
        x: -0.441163233876
        y: 0.0728935733359
        z: 0.202584217
      orientation: 
        x: -0.0474131629629
        y: -0.823935342842
        z: -0.0416467497592
        w: 0.563159028187
---
```
But in the rviz, there isn't route. And the output is attached below. Is there anything wrong?
[image]