KumarRobotics/msckf_vio #5: undefined symbol: SuiteSparse_time

pxlong, 2018-01-10:
Hi, thanks for sharing this great repo.

My development env is Ubuntu 16.04 + Ros Kinetic, and I have installed suitesparse through `sudo apt-get install libsuitesparse-dev`. However, I got the following error when I test the algorithm on euroc dataset:

> 
/opt/ros/kinetic/lib/nodelet/nodelet: symbol lookup error: /home/pxlong/ws/catkin_ws/devel/lib/libmsckf_vio.so: undefined symbol: SuiteSparse_time

kartikmohta, 2018-01-10:
Odd that we haven't seen this on any of our machines. Can you test again with the latest commits?

pxlong, 2018-01-10:
@kartikmohta thanks for the quick reply.
the error changes to 
`
[ INFO] [1515561095.065854467]: ===========================================
[ INFO] [1515561095.065862146]: Finish loading ROS parameters...
[ INFO] [1515561095.079800608]: Finish creating ROS IO...
[ INFO] [1515561095.105038079]: Finish creating ROS IO...
[ INFO] [1515561099.904323003]:  candidates: 60; track: 60; match: 58; ransac: 44/60=0.733333
[ INFO] [1515561100.423203380]:  candidates: 32; track: 32; match: 29; ransac: 21/32=0.656250
[ INFO] [1515561100.960307987]:  candidates: 51; track: 51; match: 51; ransac: 48/51=0.941176
[ INFO] [1515561101.465478163]:  candidates: 59; track: 57; match: 57; ransac: 52/59=0.881356
/opt/ros/kinetic/lib/nodelet/nodelet: symbol lookup error: /home/pxlong/ws/catkin_ws/devel/lib/libmsckf_vio.so: undefined symbol: dlarfg_
[firefly_sbx/vio-2] process has died [pid 12239, exit code 127, cmd /opt/ros/kinetic/lib/nodelet/nodelet standalone msckf_vio/MsckfVioNodelet ~imu:=/imu0 ~features:=image_processor/features __name:=vio __log:=/home/pxlong/.ros/log/5207dfe2-f5c3-11e7-b364-7085c22aa66f/firefly_sbx-vio-2.log].
log file: /home/pxlong/.ros/log/5207dfe2-f5c3-11e7-b364-7085c22aa66f/firefly_sbx-vio-2*.log
[ INFO] [1515561102.020115159]:  candidates: 37; track: 37; match: 35; ransac: 30/37=0.810811
[ INFO] [1515561102.561459653]:  candidates: 49; track: 49; match: 46; ransac: 41/49=0.836735
[ INFO] [1515561103.073954144]:  candidates: 64; track: 64; match: 62; ransac: 62/64=0.968750
[ INFO] [1515561103.613280049]:  candidates: 54; track: 53; match: 47; ransac: 38/54=0.703704
[ INFO] [1515561104.164772996]:  candidates: 32; track: 32; match: 31; ransac: 22/32=0.687500
[ INFO] [1515561104.723713336]:  candidates: 62; track: 60; match: 57; ransac: 47/62=0.758064
`

kartikmohta, 2018-01-10:
Can you provide the output of `ldd /usr/lib/x86_64-linux-gnu/libspqr.so`? It seems like you have a some issues in the installation of suitesparse.

pxlong, 2018-01-10:
Hi Kartik, I think I find the problem which causes the suitesparse error. 

I put this repo along with the **kalibr** package in the same catkin workspace. **kalibr** will build suitesparse by itself, it will lead suitesparse collisions. After moving the **kalibr** from the catkin ws, everything is OK. 
Thanks.

goldbattle, 2018-01-11:
Building on 14.04 ros indigo I had the same `undefined symbol: dlarfg_` error and was able to fix it by using the suitesparse find file from the [ceres solver repo](https://github.com/ceres-solver/ceres-solver/blob/master/cmake/FindSuiteSparse.cmake).

https://gist.github.com/goldbattle/e3a7c4a8ec751909f1c37a66d1eacf9e/revisions

kartikmohta, 2018-01-14:
@goldbattle Thanks for the fix, I've added the FindSuiteSparse.cmake from ceres-solver to the package.

closed 2018-01-15 by ke-sun