KumarRobotics/msckf_vio #6: how to get good performance on euroc dataset

pxlong, 2018-01-10:
Hi, I am running this method on euroc dataset. I follow the steps as below:
`rviz -d rviz_euroc_config.rviz`
`roslaunch msckf_vio msckf_vio_euroc.launch`
`rosbag play V1_02_medium.bag`

But the performance is not good, as you can see here: https://youtu.be/VqYoSveIdI4. This result is not comparable to https://www.youtube.com/watch?v=OXSB8Bze0cY. 

Can you please give some hints to get better performance?

kartikmohta, 2018-01-10:
Did you compile the package in Release mode? The standard `catkin_make` does not enable optimizations which leads to bad performance. Use `catkin_make -DCMAKE_BUILD_TYPE=Release`.

pxlong, 2018-01-10:
Thanks, Kartik. It works well after compiling in Release mode.  

Sorry, I did not notice that. But I am confused about the `optimizations`, is it an optimization algorithm in the code?

pxlong, 2018-01-10:
Hi Kartik, I suggest to set the default **CMAKE_BUILD_TYPE** flag to **Release** in the CMakeLists.txt with: 
`if(NOT CMAKE_BUILD_TYPE)
    SET(CMAKE_BUILD_TYPE Release)
endif(NOT CMAKE_BUILD_TYPE)
`

cumtchenchang, 2018-01-10:
@pxlong  I followed your setps, but I cannot get the map.
[ WARN] [1515588182.371265541]: MessageFilter [target=map ]: Dropped 100.00% of messages so far. Please turn the [ros.rviz.message_notifier] rosconsole logger to DEBUG for more information.
[image]

ke-sun, 2018-01-10:
@cumtchenchang 

The `fixed_frame_id` is set to `world` by default in the `msckf_vio_euroc.launch`. You can either change the launch file or change the fixed frame in rviz to fix this problem. Or more conveniently, you can use the rviz configuration files in the `rviz` folder.

cumtchenchang, 2018-01-11:
@ke-sun 
Thank you for your help.
I read your paper, how to change your cold to output dataset  trajectory, I want to compare your method with our lab method.

SkyRiderMike, 2018-01-12:
I load the configuration file in rviz folder for euroc dataset, but no trajectory show. However, it has some front-end feature tracking in the image window

closed 2018-01-16 by ke-sun