KumarRobotics/msckf_vio #7: Test msckf_vio using DUO 3D camera

YacongWang, 2018-01-11:
I have tested the msckf_vio using DUO 3D camera with the[ DUO Driver](https://github.com/duo3d/duo3d_driver), but the output on the screen just ```[ WARN] [1515654750.049876867]: Images at [3.256400] seems unsynced...```. And I also used the dataset on the [duo_vio](https://github.com/ethz-ait/duo_vio), [dataset](http://ait.inf.ethz.ch/projects/2016/duo-vio/downloads/DUOVioData.zip), but the consequence is the same. Can someone solve my problem? Thank you very much.
BTW, the extrinsic parameters of cam and IMU are that theoretically orientation is [-1,1,-1], translation is  (-6.426mm, 7.645mm, 8mm).

ke-sun, 2018-01-11:
The reason of the warning is that the stereo matching fails in the `image_processor` node. The possible reasons are:
1. The stereo images are unsynchronized. However, since the stereo images are from the DUO sensor, they are probably well synchronized.
2. The calibration, which can be either stereo calibration or cam-imu calibration, is off. Maybe you can try using the Kalibr to calibrate both the stereo cameras and the extrinsics between the cameras and IMU. Then provide the camchain-imucam.yaml file to the filter with proper modifications as listed in the example yaml files in the `config` folder.

YacongWang, 2018-01-12:
@ke-sun Thanks for your help. I have calibrated the DUO and provide the correct the parameters. But the msckf_vio cannot run, and the debug stereo window shows the images doesn't have feature extraction, and the bash output ```Images at [3.256400] seems unsynced...```. I have tested the own datasets and the calibration yaml basing the VINS-mono with the correct results.
Is there anything wrong?
[kumar_msckf_config.zip](https://github.com/KumarRobotics/msckf_vio/files/1624692/kumar_msckf_config.zip)

YacongWang, 2018-01-12:
@ke-sun Thanks. The calibration parameters of ```T_cam_imu``` is critical for software. And the translation vector from kalibr calibration tool also has incorrect results, so we should add the theoretical value or measurement by hand to instead the kalibr yaml. But the orientation vector is correct, we can use it directly.
Now, I have another problem 
```
[ INFO] [1515735555.492604003]:  candidates: 62; track: 62; match: 61; ransac: 61/62=0.983871
delta velocity: 0.691907
delta position: 1.189384
[ WARN] [1515735555.887611202]: Update change is too large.
[ INFO] [1515735556.012766042]:  candidates: 67; track: 66; match: 66; ransac: 66/67=0.985074
[ INFO] [1515735556.513728697]:  candidates: 69; track: 69; match: 69; ransac: 66/69=0.956522
[ INFO] [1515735557.030330057]:  candidates: 69; track: 69; match: 69; ransac: 69/69=1.000000
[ INFO] [1515735557.541209149]:  candidates: 66; track: 65; match: 65; ransac: 65/66=0.984848
[ WARN] [1515735557.805870685]: Start 7 online reset procedure...
[ INFO] [1515735557.808744614]: Stardard deviation in xyz: 8.051926, 4.405825, 5.769122
[ WARN] [1515735557.834342093]: 7 online reset complete...
```
the trajectory result is ```10521.406m path length```.
What should I do now to have correct the trajectory?

pxlong, 2018-01-12:
I am also facing the problem,  the image_processor always output "Images at ... seems unsynced".

ke-sun, 2018-01-12:
First I have to correct a mistake I made in the last reply. The `T_cam_imu` parameter in the launch file is actually the transformation which takes a vector in the IMU frame to the camera frame. The `T_cn_cnm1` is the transformation which takes a vector from cam0 frame to the cam1 frame. The readme is updated to address these.

I have also modified the code a bit so that you can maually set the `cam0/T_cam_imu`. Note that you don't have to change `cam1/T_cam_imu` accordingly since the filter does not use that. But make sure `cam1/T_cn_cmn1` is the correct calibration results.

Also, if you want to use the manually set cam-imu extrinsics, you may want to enlarge the initial noise  for `initial_covariance/extrinsic_rotation_cov` and `initial_covariance/extrinsic_translation_cov` in `msckf_vio_*.launch` which would allow the filter to perform online extrinsic calibration. See `config/camchain-imucam-euroc-noextrinsics.yaml` and `launch/msckf_vio_euroc_noextrinsics.launch` as an example.

However, for the stereo calibration, including camera intrinsics, distortion, and the extrinsics between the two cameras `cam1/T_cn_cnm1`, you have to use a calibration software. Manually set parameters for these entries will not be accurate enough.

YacongWang, 2018-01-14:
Thanks @ke-sun. The update repo has solved my problem. I use the ```-noextrinsics```version to run my datasets, and the results are correct with a bit drift. Maybe I need to refine the parameters. I want to know how to avoid or correct the ```online reset procedure```. In case of the large uncertainty of positions, what parameters should be modified?  And what's mean ```Total processing time```? In case of ```processing_time > 1.0/frame_rate```, what will happen.
Finally, I want to plot the trajectory on the scene image, what tool or repo I can use.

The tips to run our own datasets are: 
First, use kalibr calibration tool and have a correct ```cam1/T_cn_cmn1``` and the ```cam0/T_cam_imu```. If you are not sure the parameters are correct, just to use the ```-noextrinsics```version with the theoretical values and enlarge ```initial_covariance/extrinsic_rotation_cov and initial_covariance/extrinsic_translation_cov```. All above errors are caused by the wrong calibration parameters.
Second,  we can use the [evo](https://github.com/MichaelGrupp/evo) tool to analyze the trajectory if you have saved the timestamps, position and orientation results.

The datasets I have collected using DUO camera are upload [here](https://drive.google.com/drive/folders/1-fS56oecMoYOu8OuKlTCDCJZZ-Nh_XNv?usp=sharing). Looking forward to your good parameters in ```launch```file.

ke-sun, 2018-01-14:
I think the `cam0/T_cam_imu` in the calibration file you provide is still wrong. The rotation matrix of `cam0/T_cam_imu` is roughly `diag([-1.0 1.0 -1.0])` which corresponds to `pi rad` roll, `0 rad` pitch and `pi rad` yaw. In other words, the IMU and camera frame should have opposite positive z-axis. In the dataset `12.31-ouside-1.bag`, the z-axis of the camera frame is pointing downwards which means the positive z-axis of the IMU frame should be pointing upwards. However, the initial readings of the IMU linear acceleration in the z-axis is negative which implies the z-axis of the IMU frame is roughly pointing downwards. Therefore, the `cam0/T_cam_imu`  incorrect. Can you do a very careful checking of parameters you specified in the calibration file?

Here are some other suggestions which will be helpful:
1. Start the sensor from a stationary state when recording a bagfile. Perhaps start the sensor on a table instead of holding it in the hand.
2. Use the raw images from the sensor instead of the rectified images. Remember to modify the parameters in the calibration file accordingly.
3. Use a high resolution image (e.g. 752x480 instead of 320x240).
4. Perform general motion (just walk naturally instead of babysitting the sensor) when recording the bagfile, especially when the extrinsics between camera and IMU is manually set.

YacongWang, 2018-02-26:
@ke-sun Thx for your help. I have found the key issue for my DUO camera. The ROS driver software has something wrong with the output images. When you use a target to move from left side to the right side, you will find the phenomenon that the left and right images should be the reversal, in other words, the left image should be the right image. At the same time the position in each image(left or right image, Pixel coordinate system) should be width=image_width - positon_width. After the modification of the driver, the msckf software can run correctly.

closed 2018-02-26 by YacongWang

jyzhong, 2019-10-16:
>  the z-axis of the camera frame is pointing downwards which means the positive z-axis of the IMU frame should
How do you decide the z-axis of camera frame?

jyzhong, 2019-10-16:
@ke-sun 
How do you decide the z-axis of camera frame?   "In the dataset 12.31-ouside-1.bag, the z-axis of the camera frame is pointing downwards "

ke-sun, 2019-10-16:
Just as the general convention, locally the camera frame looks like
https://johnloomis.org/ece564/notes/CameraCalibration/2012_0524/fig5.png

However, this does not prevent the z-axis of the camera frame pointing down.

jyzhong, 2019-10-17:
> Just as the general convention, locally the camera frame looks like
> https://johnloomis.org/ece564/notes/CameraCalibration/2012_0524/fig5.png
> 
> However, this does not prevent the z-axis of the camera frame pointing down.

@ke-sun thanks for your rapid reply. It really means a lot. BTW, i wanna hope you can do me a favor. I cann't get a good kalibr result. i find the initial timeshift between imu and camera is bigger.  Why ? In fact, this is less imu losed when i record the bag. Hope get your reply. Thanks.