KumarRobotics/msckf_vio #13: trajectory

jixiedaima, 2018-01-20:
I can't show the trajectory information in rviz and I don't know why?
platform: ubuntu16.04.2 kinetic opencv3.3.1

ke-sun, 2018-01-20:
The software does not publish the trajectory as `nav_msgs/Path`.

jixiedaima, 2018-01-21:
How can I get the trajectory information?

kartikmohta, 2018-01-21:
You can write a simple node to accumulate the output odometry messages into a `nav_msgs/Path` message for visualization.

jessicajie, 2018-01-22:
Can you solve it ? Can you explain it in detail？thanks a lot .

kartikmohta, 2018-02-05:
A simple node to take `nav_msgs/Odometry` messages and populate a `nav_msgs/Path` from it: https://gist.github.com/kartikmohta/67cafc968ba5146bc6dcaf721e61b914

You can remap the topics to match whatever your input and output topic names are.

SouthEastleo, 2018-02-05:
Thanks a lot，I have solved it.Best wishes for you

TurtleZhong, 2018-02-05:
[image]

TurtleZhong, 2018-02-05:
Hello, I save the trajectory and plot the trajectory. But I found that the translation is really strange. Below are the trajectory of MH_01_easy.bag.
[image]
I only add these in your code:

```
  Vector3d t_b_w =  T_b_w.translation();
  outFile << t_b_w(0) / 1000. << " " << t_b_w(1) / 1000. << " " << t_b_w(2) / 1000. << endl;
```
and the plot the trajectory using this:
```
import numpy as np
import matplotlib.pyplot as plt

trajectory_path = 'trajectory.txt'
trajectory_data = np.loadtxt(trajectory_path, dtype=np.float64, delimiter=' ')
gt_path = '../gt_path/mh_01_easy.csv'
gt_data = np.loadtxt(gt_path, dtype=np.float64, comments='#', delimiter=',')
gt_data = gt_data[:, 1:4]

ax = plt.subplot(1, 2, 1, projection='3d')
ax.plot(gt_data[:, 0], gt_data[:, 1], gt_data[:, 2],
            '-', alpha=0.5, color="red", )

ax = plt.subplot(1, 2, 2, projection='3d')
ax.plot(trajectory_data[:, 0], trajectory_data[:, 1], trajectory_data[:, 2],
            '-', alpha=0.5, color="blue", )

plt.show()
```
I didn't know why the translation is so strange.
Can anyone help me?
@jixiedaima @ke-sun @kartikmohta @jessicajie @SouthEastleo 
Thank you!

ke-sun, 2018-02-05:
The software does not run on EuRoC `MH_01_easy.bag` and `MH_02_easy.bag`. As explained in the README, the algorithm requires the sensor to start from staic in order to initialize the orientation and IMU bias. unfortunately, `MH_01_easy.bag` and `MH_02_easy.bag`  do not have the initial static period.

jixiedaima, 2018-02-06:
Can it get good performance in other EuRoC ?

ke-sun, 2018-02-06:
The rest should be fine.

TurtleZhong, 2018-02-06:
@ke-sun Really thank you！！！

chutsu, 2018-02-12:
@TurtleZhong can you quickly comment why you divided the translations by 1000?

alexSongB, 2018-03-09:
I want to know how long is the initial static period?Thank you.

ke-sun, 2018-03-09:
It depends on the IMU rate. The software currently needs 200 IMU msgs to estimate the initial orientation. But you can change it to other reasonable numbers. See
https://github.com/KumarRobotics/msckf_vio/blob/1e524eb957b9ea32350dfdc9c88abf56563e4680/src/msckf_vio.cpp#L239

alexSongB, 2018-04-10:
I get something confused. Can you quickly comment why you divided the translations by 1000 in your code snippet? what‘s the unit for the t_b_w(0) ? @ke-sun

alexSongB, 2018-04-10:
I want to know how can you get the error between groundtruth and trajectory?where is the groundtruth? I am looking forward for your reply, thank you.

ke-sun, 2018-04-10:
There is ground truth in the EuRoC dataset. You may want to compare with that. Regarding #19 , the path messages are only for debugging purpose, they are not included in the current version.

alexSongB, 2018-04-11:
Is the ground truth included in data.csv .but I found that the rows of trajectory is more less than the row of the data.csv. So I can't compare the trajectory and ground true by different sharp.

yuyadanyadan, 2018-04-13:
Did someone write the node???  Tank you very much !

chenwinki, 2018-07-03:
I know a tool named EVO used for compare trajectory

sjlszz, 2019-08-20:
> 
> 
> Thanks a lot，I have solved it.Best wishes for you

can u show me the detail of the change?thank u

sjlszz, 2019-08-21:
> A simple node to take `nav_msgs/Odometry` messages and populate a `nav_msgs/Path` from it: https://gist.github.com/kartikmohta/67cafc968ba5146bc6dcaf721e61b914
> 
> You can remap the topics to match whatever your input and output topic names are.

     

can u show me the detail of the change?thank u

closed 2021-02-02 by ke-sun