KumarRobotics/msckf_vio #17: Question regarding the Nullspace Trick

chutsu, 2018-01-29:
In calculating the filter residuals:

```
   r = z - z_hat = H_x * X_err + H_f * p_f + n 
   # wish I could type latex in github
````
As per the paper, you're following Mourikis's Null Space trick to remove the correlation of `X` (camera pose state) and `p_f` (feature position) by projecting the above to the left null space of `H_f`.

In the code ([here](https://github.com/KumarRobotics/msckf_vio/blob/master/src/msckf_vio.cpp#L906)) or this snippet:
```
// Project the residual and Jacobians onto the nullspace
// of H_fj.
JacobiSVD<MatrixXd> svd_helper(H_fj, ComputeFullU | ComputeThinV);
MatrixXd A = svd_helper.matrixU().rightCols(jacobian_row_size - 3);

H_x = A.transpose() * H_xj;
r = A.transpose() * r_j;
```

 I was curious to find you projected the residuals `r` and `H_x` but you didn't project the noise `n`, is there a reason for not doing so?

ke-sun, 2018-01-29:
The numerical values for the diagonal elements of the noise covariance matrices (before and after projected to the null space) does not change. It is the dimension of the covariance matrix changes. However, the change of dimension can be easily handled during the measurement update step, which does not need any computation beforehand.

chutsu, 2018-02-07:
Got it, thanks.

closed 2018-02-07 by chutsu