KumarRobotics/msckf_vio #20: run error for help

YJCITA, 2018-03-07:
I compile msckf on unbuntu 16.04,the compile is ok, but when I run the: 
roslaunch /home/yj/src/ros_ws/src/msckf_vio/launch/image_processor_fla.launch
there is error info like this:
process[fla/image_processor-1]: started with pid [3649]
type is msckf_vio/ImageProcessorNodelet
[ERROR] [1520426217.959420841]: Failed to load nodelet [/fla/image_processor] of type [msckf_vio/ImageProcessorNodelet] even after refreshing the cache: MultiLibraryClassLoader: Could not create object of class type msckf_vio::ImageProcessorNodelet as no factory exists for it. Make sure that the library exists and was explicitly loaded through MultiLibraryClassLoader::loadLibrary()
[ERROR] [1520426217.959466473]: The error before refreshing the cache was: Failed to load library /home/yj/src/ros_ws/devel/lib//libimage_processor_nodelet.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library (Poco exception = /home/yj/src/ros_ws/devel/lib//libimage_processor_nodelet.so: undefined symbol: _ZN12class_loader20class_loader_private30getCurrentlyLoadingLibraryNameEv)
[fla/image_processor-1] process has died [pid 3649, exit code 255, cmd /opt/ros/kinetic/lib/nodelet/nodelet standalone msckf_vio/ImageProcessorNodelet ~imu:=sync/imu/imu ~cam0_image:=sync/cam0/image_raw ~cam1_image:=sync/cam1/image_raw __name:=image_processor __log:=/home/yj/.ros/log/27874b36-21ec-11e8-ada2-54e1ade5394d/fla-image_processor-1.log].
log file: /home/yj/.ros/log/27874b36-21ec-11e8-ada2-54e1ade5394d/fla-image_processor-1*.log
[
[image]
](url)

alexSongB, 2018-03-08:
you should run roslaunch msckf_vio msckf_vio_euroc.launch

kartikmohta, 2018-03-08:
@YJCITA The actual error is:
```
Could not load library (Poco exception = /home/yj/src/ros_ws/devel/lib//libimage_processor_nodelet.so: undefined symbol: _ZN12class_loader20class_loader_private30getCurrentlyLoadingLibraryNameEv)
```
which seems a bit odd. Can you clean your workspace (remove the `build` and `devel` folder), recompile and try again?

YJCITA, 2018-03-14:
@kartikmohta I have clean your workspace (remove the build and devel folder), recompile and try again, but the result is the same?
[image]

YJCITA, 2018-03-14:
@alexSongB  I run "roslaunch msckf_vio image_processor_fla.launch", but it still failed.
[image]

YJCITA, 2018-03-14:
@alexSongB @kartikmohta  
My gcc version is:gcc (Ubuntu 4.9.3-13ubuntu2) 4.9.3, is this ok?

moshanATucsd, 2018-08-15:
I encountered a similar error and solved it using `source devel/setup.bash`

closed 2021-02-02 by ke-sun