KumarRobotics/msckf_vio #21: catkin_make & catkin_make --pkg msckf_vio --cmake-args -DCMAKE_BUILD_TYPE=Release

gyjun0230, 2018-03-09:
Hello , i got a strange problem. I build the msckf_vio package with catkin_make, the VIO result being very poor. However , if I build the package  with it（catkin_make --pkg msckf_vio --cmake-args -DCMAKE_BUILD_TYPE=Release）， the result is good. so , that's why???

kartikmohta, 2018-03-09:
Without `--cmake-args -DCMAKE_BUILD_TYPE=Release`, it builds without compiler optimizations which leads to bad performance. For any benchmarks, you should build in `Release` mode.

gyjun0230, 2018-03-09:
Thank you very much for replying me. The program is very well and grate you to share for everyone to use.
I only understand the use of debug or release affect the speed of the program, but the results should not be  affected. I'm very curious. Thank you very much!

alexSongB, 2018-03-15:
Hello, Do you run the program in Euroc datasets or device? @gyjun0230

biendltb, 2020-08-19:
I faced the same issue with the build flag and also wondered how it relates to the compiler optimization. How could we detect if the accuracy is affected by the performance since there are no obvious clues when it happens.

kartikmohta, 2020-08-24:
The main problem when compiling without optimizations is that the computation takes a long time for every frame so if you play a bag at normal speed, a lot of frames might be skipped which would lead to bad accuracy.

biendltb, 2020-08-24:
Thanks for the reply, @kartikmohta.
I think that the issue might also come from a mechanism in the algorithm or one of its dependent libraries since I didn't expect the result is at that poor. I could see the algorithm run slightly slower but a very dramatic drift occurs even with the slightest movement of the camera. Let's say if I feed a stream of data at 10 FPS and the algorithm can only handle 7 FPS, the drift is still much worse than playing a 7-FPS rosbag.
I'm trying to understand deeper about the error caused by this build flag.

kartikmohta, 2020-08-28:
Without optimizations the algorithm will run much slower (around 10 times is not unexpected), so you can try playing the bag at a much slower rate and see if the drift is less.

closed 2021-02-02 by ke-sun