KumarRobotics/msckf_vio #24: Trajectory is not smooth

shishiteng, 2018-03-21:
Hi,Sun.
I have tested msckf_vio and it worked very well.But one thing is strange that the trajectory always has some jagged jitter even I walk straight.How does this jitter come from?What can I do to eliminate it?

tomas789, 2018-03-21:
Such a behavior is normal and expected. How does MSCKF work is that you are propagating IMU data and in this process, you are accumulating an error. When a new image arrives, it estimates the amount of drift accumulated (technically not true, it just minimizes the reprojection error) and updates the state such that the accumulated drift if (hopefully) smaller. Those updates are where the jitter comes from.
There is not much you can do about it. To suppress this behavior, you can either make the filter better (which is very complicated) or introduce some sort of trajectory smoothing. 5th order splines are especially good for this purpose. Keep in mind that it would inevitably lower the accuracy of the filter.

ke-sun, 2018-03-21:
@shishiteng 
I think @tomas789 's reply explains most of it. One more interesting thing to note is that the motion of camera/IMU determines the observability of the filter, since it is a nonlinear system. In the case of straight line translation without rotation, the filter loses the observability on the acc bias. Then the uncertainty (covariance) of the filter will grow rapidly (the high speed flight video with the paper is a nice example). Once the prior uncertainty is large, the following estimation from the filter tends to be jumpy.

shishiteng, 2018-03-22:
Thanks for your quick replies @ke-sun @tomas789 .
Can I understand what you mean this way? When the filter gets new measurements, it will try to reduce the cumulative error. If the previous error accumulates too much, this update will cause a large change of camera pose.
Is there any way to slow down the accumulation of errors? Can I get it by adding more frames in the sliding window ,or reducing feature nosie of image, or any other way?

tomas789, 2018-03-22:
There certainly are things that can change this behaviour. Just keep in mind, that it will, to some extent, always be there. 
You can try optimizing:
Feature noise parameter
Size of sliding window
A schema pro pruning of the poses from the sliding window
Number of features in each cell in image processor node. 
Size of grid in image processor node
IMU random walk parameters
Image resolution (you would get more features for residualisation)

shishiteng, 2018-03-23:
@tomas789 I found that the trajectory smoothed a lot after the feature noise was turned down, but the accumulated error became larger. How should this parameter be set, is there any way to calibrate it?

closed 2021-02-02 by ke-sun