KumarRobotics/msckf_vio #25: how to run msckf_vio?

Jhon82, 2018-03-22:
thanks very much for this project. 
I have successfully build using "catkin_make --pkg msckf_vio --cmake-args -DCMAKE_BUILD_TYPE=Release", the thing is how could I run this program?
some one mentioned " roslaunch msckf_vio msckf_vio_euroc.launch" , but where could I find this "msckf_vio"?  which path should I add to the ROS_PACKAGE_PATH?
ps: I'm a totally newbie about this ros thing.

alexSongB, 2018-03-26:
First you need to run "$source devel/setup.bash" in your workspace. I wish it will help you.

chutsu, 2018-03-28:
Added a little bit more instructions in the README, if you're still struggling please make sure you go through the ROS tutorials first. [link](http://wiki.ros.org/ROS/Tutorials)

hhmaizi, 2018-03-30:
thanks a lot, I will try.

closed 2021-02-02 by ke-sun