KumarRobotics/msckf_vio #26: generateInitialGuess?

qcz1107, 2018-03-24:
void Feature::generateInitialGuess(
    const Eigen::Isometry3d& T_c1_c2, const Eigen::Vector2d& z1,
    const Eigen::Vector2d& z2, Eigen::Vector3d& p) const {
  // Construct a least square problem to solve the depth.
  Eigen::Vector3d m = T_c1_c2.linear() * Eigen::Vector3d(z1(0), z1(1), 1.0);

  Eigen::Vector2d A(0.0, 0.0);
  A(0) = m(0) - z2(0)*m(2); 
  A(1) = m(1) - z2(1)*m(2);

  Eigen::Vector2d b(0.0, 0.0);
  b(0) = z2(0)*T_c1_c2.translation()(2) - T_c1_c2.translation()(0);
  b(1) = z2(1)*T_c1_c2.translation()(2) - T_c1_c2.translation()(1);

  // Solve for the depth.
  double depth = (A.transpose() * A).inverse() * A.transpose() * b;
  p(0) = z1(0) * depth;
  p(1) = z1(1) * depth;
  p(2) = depth;
  return;
}

Isn't that ?
  Eigen::Vector2d A(0.0, 0.0);
  A(0) = m(0)/m(2) - z2(0); 
  A(1) = m(1)/m(2) - z2(1);

ke-sun, 2018-03-26:
Can you show the math leading to your conclusion.
Based on my calculation, I think the equation in the code is correct.

qcz1107, 2018-03-26:
From the implementation in the code, may not fully catch the purpose, I think A is the disparity distance in the camera normalised plane. While z2(0,1,2) is at the normalised plane (z=1), but m(0,1,2) isn't, so that I intend to normalise m to make m(2) = 1. And that will provide similar triangle and A/1=b/d, so that Ad=b.

ke-sun, 2018-03-27:
In case you missed, "m" in the equation is not the 3-D coordinates of the feature in the c2 frame.

qcz1107, 2018-03-27:
Thx. I thought the m is in the normalise plane in c2. Will check Stergios's paper.

chenwinki, 2018-06-18:
Stergios's paper???  Could you give me the complete name of this paper talk the equations in the void Feature::generateInitialGuess?  thank you

qcz1107, 2018-06-18:
https://ieeexplore.ieee.org/document/4209642/

The estimation equation is correct and aligned with the paper. However, I did not figure out what is wrong with my geometry derive which drive out my question. Will figure out that when i have time.

qcz1107, 2018-06-18:
@chenwinki see above

symao, 2018-08-08:
The basic constraint: R(u1,v1,1)'d+t // (u2,v2,1)', which means the 3D point transformed from cam1 to cam2, should be parallel to the observation in cam2 normalized plane. R,t is the transformation from cam1 to cam2, d is depth.

Use the cross product property: a//b <=> axb=0, the constrain can be write as (R(u1,v1,1)'d + t) x (u2,v2,1)' = 0 
=> R(u1,v1,1)' x (u2,v2,1)'  d = -t x (u2,v2,1)'

qcz1107, 2018-08-08:
I do not think we need to use the epipolar constrain this way, but, that is a way to compute.

closed 2018-08-08 by qcz1107

moshanATucsd, 2019-03-27:
Hi, I have noticed that there are only 2 dimensions considered in solving the least squares. I am wondering why we do not consider the third dimension here? 

```
  Eigen::Vector2d A(0.0, 0.0);
  A(0) = m(0) - z2(0)*m(2);
  A(1) = m(1) - z2(1)*m(2);

  Eigen::Vector2d b(0.0, 0.0);
  b(0) = z2(0)*T_c1_c2.translation()(2) - T_c1_c2.translation()(0);
  b(1) = z2(1)*T_c1_c2.translation()(2) - T_c1_c2.translation()(1);
```