KumarRobotics/msckf_vio #30: question about MsckfVio::predictNewState

PetWorm, 2018-04-04:
In '// Some pre-calculation':
Why multiply a 'cos(gyro_norm*dt*0.5)' term when gyro_norm is no bigger than 1e-5 ?
And any references for these imu propogation methods ? I read the paper and it just said "we  apply  a
4th order  Runge-Kutta  numerical  integration  of Eq.  (1)  to  propagate  the  estimated  IMU  state."
Thanks a lot.

chutsu, 2018-04-04:
They are using the equations found in ["Indirect Kalman Filter for 3D Attitude Estimation"](http://mars.cs.umn.edu/tr/reports/Trawny05b.pdf) Section 1.6.1 Zeroth Order Quaternion Integrator. Here they are pre-computing the aforementioned section to integrate the quaternion before hand to use the Runge-Kutta later.

The equations they implemented are Eq. 101 and 103 (for very small angular velocities, due to numerical instabilities). But I'n not sure the `cos(gyro_norm*dt*0.5)` was needed?

ke-sun, 2018-04-06:
I agree the `cos` term can be approximated with 1 when the angle is no greater than 1e-5. However, it just depends on to what level one what to approximate. The only problematic term for the small angle is `1/|w|`. Therefore, only the second term is converted using L'Hopital's rule.