KumarRobotics/msckf_vio #31: Initialization question

jian-li, 2018-04-06:
Hi, Thank you for sharing the code. I have tested the code on euroc and it works perfect like a charm. But I have a problem about vio initialization. In your code, the static initialization process give the initial state, but I don't found the process to process the static scheme after vio initialization. In my opinion, the imu continiously propogates the system state, but in the static scheme, the vision part has no disparity, so the msckf visual update should never be performed. Could you please explain the question? Thx! :)

ke-sun, 2018-04-06:
I don't quite understand this part, "but I don't found the process to process the static scheme after vio initialization", in the question. But it is a stereo vio. The disparity can be recovered even the robot is static.

jian-li, 2018-04-07:
Thank you for you quick reply! :) I read the code again and got it. The stereo features are set to be initialized when they are detected,  when the sliding window is full, two frames are dropped and their stereo features can still be used to do msckf update even when the robot is static.

closed 2021-02-02 by ke-sun