KumarRobotics/msckf_vio #32: about state augmentation jacobian matrix

poaongithub, 2018-04-11:
hi, I'm confused about the jacobian matrix Ji provied by your code. j block related to q_c_i  and i_P_c:

`  J.block<3, 3>(3, 12) = Matrix3d::Identity();
  J.block<3, 3>(3, 18) = Matrix3d::Identity();`

could you please tell me how to derive this? any reference paper? thanks a lot.

ke-sun, 2018-04-12:
It should be in the Appendix of our Paper.

poaongithub, 2018-04-13:
thanks for getting back to me, sorry, my bad, let me more specific, 
1. Ji<3,3>(0,3) is defined as  "-C(q_G_I)' * [P_C_I]x" in your guys paper, but in your code :
`J.block<3, 3>(3, 0) = skewSymmetric(R_w_i.transpose()*t_c_i);`
same as the Ji of A. I. Mourikis' paper.

2. when I try to derive Ji, I get:
D(q_C_I x q_G_I) / D(theta_C_I) = R_G_I'
D(P_I_G + C(q_G_I))' * P_C_I) / D(P_C_I) = C(q_G_I))'

 I don't know whether it's correct and where is the mistake if it's wrong. Please tell me, Thanks!
@ke-sun

tomas789, 2018-06-05:
I also noted this discrepancy. According to my own calculations, the formula reported in the paper is correct. For some reason, when I try to implement it, the results get worse. My guess that it would be that @ke-sun experienced similar results and for this reason, they left it there. I cannot explain why it even works with the formula from Mourikis's paper. Those two formulas are not equal.

Ajithkumar1, 2019-05-06:
Hi @ke-sun 

Have you experienced the same result when using formula from the paper as noted by @tomas789  ?.
My derivations have also landed me to the equations given in paper. 
But, the results were worse when I tried implementing it. 

Can you please explain about this discrepancy?

closed 2021-02-02 by ke-sun