KumarRobotics/msckf_vio #33: two-point ransac algorithm mathematical derivation

jian-li, 2018-04-18:
Hi, ke! Could you please supply some materials about the two-point ransac algorithm mathematical derivation. Thank you very much!

jian-li, 2018-04-18:
Thx, I got it! Just use the epipolar constraint!

jian-li, 2018-04-18:
Hi, ke! Another question, why the Sampson’s error is not used for outlier curling? In the code, the error is defined as `VectorXd error = coeff_t * model;`, but in MVG and some paper the error should by divided by sqrt(||Ex||^2+||x'E||^2), could you please explain it? Thx:)

chutsu, 2018-04-20:
They are using the 2 point ransac from ["2-Point-based Outlier Rejection for Camera-IMU Systems with
applications to Micro Aerial Vehicles" by Chiara Troiani](https://www.ifi.uzh.ch/dam/jcr:5030f1bb-dcbd-44a9-a417-f12cc589c00a/ICRA14_Troiani.pdf) I would start there.

w1019220367, 2018-06-13:
@jian-li hi, I still donot understand this part. Can you give me some reference papers you have read? thx.

symao, 2018-07-09:
@wk199 Suppose the features in previous and current normalized plane are p_1, p_2, the rotation R and translation t, then, Rp_1 , p_2 and t are co-plane, thus (Rp_1 x p_2)t = 0. Since there is scale factor in t, set one of (tx,ty,tz) to 1, so you can random choose two points and formulate two constraint equations to solve t.

struggleforbetter, 2018-11-15:
> @wk199 Suppose the features in previous and current normalized plane are p_1, p_2, the rotation R and translation t, then, Rp_1 , p_2 and t are co-plane, thus (Rp_1 x p_2)t = 0. Since there is scale [...]

if  the points are not normalized after taking rotation , Rp_1 , p_2 and t are co-plane. But the function rescalePoints() changes the values of Rp_1,p_2, in only x and y coordinate,while z is constant to 1. Are they still co-plane? I am really confused. Thanks for answer.

struggleforbetter, 2018-11-15:
suppose p_1,P_2 are  the features in previous and current normalized plane .they are satisfied with : 
p_2^T * [t]× * R * p_1 = 0;when taking matrix H = (s,0,0;0,s,0;0,0,1) to p_1,p_2,then p_1' = H*p_1,p_2' = H*p_2. does  a R' ,t' exist to p_2'^T *[t']× * R' * p_1' = 0? that is ,H*[t]× * R *H == [t']× * R' ? sorry for my poor English,sorry @jian-li @symao

struggleforbetter, 2018-11-15:
$p_1$

struggleforbetter, 2018-11-16:
suppose:
t:(1,1,1)^T
R:I（Identity matrix）
x2：（1/2,0,1)^T
x1：（0,-1,1)^T
x2^T * [t]x * R *x1 == 0

now suppose scale_factor = 2
x2=>x2' ,x1=>x1' (only change x,y coordinate)
x2':(1,0,1)
x1':(0,-2,1)
now，x2'^T * [t]× * x1'  /= 0 @jian-li @wk199 @symao

realdealneil, 2018-12-07:
I'm familiar with this paper on 2-point RANSAC:

https://hal.inria.fr/hal-01024890/document

but what is implemented in UPenn's MSCKF is different (though related, since it involves the epipolar constraint equation).  Has this particular 2-point RANSAC algorithm been published elsewhere?

closed 2021-02-02 by ke-sun