KumarRobotics/msckf_vio #34: run rviz  appear error

jessicajie, 2018-04-21:
$ roslaunch msckf_vio msckf_vio_euroc.launch 
$  rosrun rviz rviz -d /home/wj/Downloads/msckf_vio_workspace/src/msckf_vio/rviz/rviz_euroc_config.rviz

[ERROR] [1524280690.236418940]: Client [/rviz_1524280132799940969] wants topic /firefly_sbx/vio/odom to have datatype/md5sum [nav_msgs/Path/6227e2b7e9cce15051f669a5e197bbf7], but our version has [nav_msgs/Odometry/cd5e73d190d741a2f92e81eda573aca7]. Dropping connection.

Global Status:Warn
Fixed Frame :   No tf data.Actual error:Fixed Frame [world] does not exist

Anyone can help me to solve this problem?Thanks a lot!

ke-sun, 2018-04-23:
It seems you want to visualize `/firefly_sbx/vio/odom` as `nav_msgs/Path`, but the message is of type `nav_msgs/Odometry`.

chenwinki, 2018-06-26:
you can write a .launch file to run rviz,the file is 
<launch> 
       
<node pkg="rviz" type="rviz" name="rviz" 
    args="-d $(find msckf_vio)/rviz/rviz_euroc_config.rviz"/> 
       
</launch>

closed 2021-02-02 by ke-sun