KumarRobotics/msckf_vio #37: JPL quaternion to Hamilton ?

442384270, 2018-05-30:
The msckf_vio use JPL quaternion. That's quite strange as both eigen and ros use Hamilton.
Is there any future plan to change JPL quaternion to Hamilton ?
Or, is there any reference material ?
Thanks.

tomas789, 2018-06-05:
Although I cannot say for the authors but what would be the advantage of conversion here? It might not be obvious at first glance but majority of the formulas would need to be derived again and debugged again. The total effort in doing so would be somehow comparable to writing the whole thing from scratch. I'd say the slight advantage of having Hamilton notation just doesn't pay off. Plus all the literature on MSKCF uses JPL notation and it might actually obfuscate things more then it would help.

442384270, 2018-06-12:
> Plus all the literature on MSKCF uses JPL notation

That sounds reasonable

closed 2018-06-12 by 442384270