KumarRobotics/msckf_vio #39: About the isometric transform in MsckfVio::publish()

yaobaishen, 2018-06-15:
I am reading the msckf code, and is confused by the transformation from T_i_w to T_b_w in MsckfVio::publish():

  Eigen::Isometry3d T_b_w = (MUState::T_imu_body) * T_i_w * (IMUState::T_imu_body.inverse());

From the literal meaning, if want to get T_b_w, we could use :  T_body_imu * T_i_w, it's the same as: (IMUState::T_imu_body.inverse()) * T_i_w.

So I can’t understand why the expression adds a left multiply with (MUState::T_imu_body)? Anyway, just one multiplication is enough, but here we see twice.

I am not quite familiar with the multiview geometry, can you show some tips? thanks.

ke-sun, 2018-06-19:
`T_imu_body` is always set to identity in the configuration file. I would suggest do not bother with this =P. 

However, if it is really your case that the IMU frame the body frame are different. What should be published is the transform from the current body frame to the initial body frame, `T_b_w`. What is estimated by the VIO is the transform from the current IMU frame to the initial IMU frame, `T_i_w`. The `w` in the two variables means initial body and initial IMU frame respectively, not a "fixed world frame". Sorry for the confusion.

yaobaishen, 2018-06-22:
Thanks for explanation!
In your comment:
"The w in the two variables means initial body and initial IMU frame respectively, not a "fixed world frame"."
In Prof.Mourikis's original MSCKF paper 2006, "the global frame is chosen as an ECEF frame". So I want to confirm, since in your implement the IMU frame and body frame is just the same, so the initial IMU/body frame is chosen as the global frame, not using ECEF, right?

ke-sun, 2018-06-22:
You are right, the global frame is not ECEF frame in the code.

closed 2021-02-02 by ke-sun