KumarRobotics/msckf_vio #41: About image exposure controller

Lwenqi, 2018-06-19:
Hi,Sun.
I read your paper regarding this msckf_vio and noticed that you mentioned you are using a fast-adapting external controller to maintain constant average image brightness. I looked through your paper references and this git repo but could not find any other information on that. Would you mind if you share what kind of algorithm you are using and if it is open sourced? Thank you.

ke-sun, 2018-06-19:
@berndpfrommer would be the best person to answer this. =)

berndpfrommer, 2018-06-20:
The auto exposure code is not open source because for one the actual implementation is ugly, and second because it ties into other software that is not open sourced.

Here a few (edited) notes that I sent to somebody who asked about how it works:

- what we implemented is super simple but works well in practice:
   new_shutter_time =  current_shutter_time * (desired_brightness/current_brightness)
 You can go fancier by implementing
  new_shutter_time = current_shutter_time * (desired_brightness/current_brightness)^alpha
 alpha <= 1, but in practice alpha = 1 works just fine.
- brightness is computed from every 16th row and column, so in fact only 1 out of every 256 is used for brightness computation.
- threshold on brightness: we only change shutter time if |desired_brightness - current_brightness| > threshold. This is so that we don't torture the ptgrey cameras with constant shutter speed changes.
- region of interest (ROI) support: we compute brighness only from the bottom X%. Typically X=70, so we will ignore the top 30% of the image, because that's often where the sun or sky are, and where we don't usually pick up features. 
- configurable max shutter limit: the max shutter time is the min of the shutter time imposed by frame rate, and a configurable hard shutter limit (in case we get motion blur and decide we'd rather live with a darker image and/or gain noise than the motion blur). 
- auto gain: once we hit max shutter, we turn on auto gain (and when it get's brighter again, we take the gain off first, then decrease shutter).

Lwenqi, 2018-06-22:
Thank you berndpfrommer and sun for your kindly help!

Lwenqi, 2018-06-26:
@berndpfrommer Kindly check if you could share the name of the software that is not open sourced?  Is it a third party driver for pointgrey camera? Thank you very much.

berndpfrommer, 2018-06-26:
We have developed a camera-imu-sync ROS node that takes the hardware-synced images + imu messages from the pointgrey cameras + vector nav and publishes them. From what I understand, that node is not in the public domain, and I don't think we have intentions to release it because of the work involved in cleaning up the code and dealing with the licensing issues.
It is that node that interacts with the the auto exposure node through ROS messages. The workings of the auto exposure node is dependent on the cam-imu-sync node, and one cannot be understood without the other.
In fact, in my opinion the auto exposure should be directly coded into the cam-imu-sync node, rather than kept as a separate node, to avoid additional latencies. We initially separated the two so it was easy to switch on and off auto exposure should we encounter frame delays and de-syncing due to changing the exposure while keeping the camera running. Now we should go back and clean everything up, and that is unfortunately quite some work.

closed 2021-02-02 by ke-sun