KumarRobotics/msckf_vio #43: Measurement Jacobian

yuweijheng, 2018-06-20:
Thanks for sharing the code. 
I am wondering why is `dpc0_dxc.leftCols(3) = skewSymmetric(p_c0)`? Shouldn't it be `-R_w_c0 * skewSymmetric(d_theta) * (p_w - t_c0_w) = R_w_c0 * skewSymmetric(p_w - t_c0_w) * d_theta`?

ke-sun, 2018-06-21:
The reason is that we use the left multiplication for the delta changes, i.e.
[image]
instead of
[image]
where $\hat{R}$ represents the estimated orientation, while $dR$ stands for the difference between the estimated and the true orientation.

yuweijheng, 2018-06-21:
Thanks for your reply!
Another question would be the entries of the matrix J in `stateAugmentation`.
The uncertainty representation for ![equation](http://latex.codecogs.com/gif.latex?\tilde{\mathbf{p}}^{w}_{c}) can be easily derived of the following form: 
![equation](http://latex.codecogs.com/gif.latex?\tilde{\mathbf{p}}^{w}_{c}=\tilde{\mathbf{p}}^{w}_{i}+\hat{\mathbf{R}}^{w}_{i}\left(\mathbf{p}^{i}_{c}\times\right)\tilde{\boldsymbol{\theta}}_{i}) 
(For some reasons, the results get worse than 
![equation](http://latex.codecogs.com/gif.latex?\tilde{\mathbf{p}}^{w}_{c}=\tilde{\mathbf{p}}^{w}_{i}+\left(\hat{\mathbf{R}}^{w}_{i}\mathbf{p}^{i}_{c}\times\right)\tilde{\boldsymbol{\theta}}_{i})
Thus, the corresponding code matches this formula as in the original MSCKF paper.)

But I am a bit confused how to derive the uncertainty representation for ![equation](http://latex.codecogs.com/gif.latex?\tilde{\boldsymbol{\theta}}_{c}=\mathbf{R}^{i}_{w}\tilde{\boldsymbol{\theta}}_{i})? I am not sure if this is a typo or similar reasons as above.  Can you briefly describe or walk me through how to derive the uncertainty representation for the orientation of camera?

Thanks.

ke-sun, 2018-06-22:
My last impression was that the two equations resulted in similar accuracy. It sounds weird the results get worse in your case.

I think the correct transformation is 
[image]
which leads to the error state relation as (assuming the cam-imu extrinsics is known),
[image]
Hopefully, you would find it makes more sense this time.

chutsu, 2018-06-26:
I could be completely wrong, but could the uncertainty representation for [image] have two different forms:

![](https://camo.githubusercontent.com/7251ec3742e0eecf24cf20ad0a50b2252efd817d/687474703a2f2f6c617465782e636f6465636f67732e636f6d2f6769662e6c617465783f25354374696c64652537422535436d617468626625374270253744253744253545253742772537445f253742632537443d25354374696c64652537422535436d617468626625374270253744253744253545253742772537445f253742692537442b2535436861742537422535436d617468626625374252253744253744253545253742772537445f253742692537442535436c6566742535436c666c6f6f722535436d617468626625374270253744253545253742692537445f2537426325374425354374696d6573253543726967687425354372666c6f6f7225354374696c6465253742253543626f6c6473796d626f6c25374225354374686574612537442537445f25374269253744)

and

![](https://camo.githubusercontent.com/6abc7a95eb1b02c8d9d71b4ef59f8e794eabbd10/687474703a2f2f6c617465782e636f6465636f67732e636f6d2f6769662e6c617465783f25354374696c64652537422535436d617468626625374270253744253744253545253742772537445f253742632537443d25354374696c64652537422535436d617468626625374270253744253744253545253742772537445f253742692537442b2535436c6566742535436c666c6f6f722535436861742537422535436d617468626625374252253744253744253545253742772537445f253742692537442535436d617468626625374270253744253545253742692537445f2537426325374425354374696d6573253543726967687425354372666c6f6f7225354374696c6465253742253543626f6c6473796d626f6c25374225354374686574612537442537445f25374269253744)

because:

1. You "right multiplied" the delta
2. Used (eye(3) + skew(\tilde{\theta})) for small angle approx as supposed to (eye(3) - skew(\tilde{\theta})) (note: the minus.)

[image]

I may have made mistakes, please be kind!

tomas789, 2018-06-27:
To answed the last question from the image, no, because of the angular error expressed in the body frame instead of the global frame. This is often called local or global angular error.

symao, 2018-08-13:
Hi, i am confused about the difference in Jacobian of state augmentation. The $dp_c/dv$ is identity in Sun's paper while zeros in Mourikis's paper. Anyone know why?

andylei77, 2019-07-28:
I think  the delta in https://github.com/KumarRobotics/msckf_vio/issues/43#issuecomment-400434979 by @chutsu is wrong. 
[image]

True answer should be:
[image]

By the way, **why delta of JPL is "-"?**
**Because JPL is  Left-handed.**
 My derivation is follow: 
[image]

closed 2021-02-02 by ke-sun