KumarRobotics/msckf_vio #57: Processing slow in Jeston TX2

arjunskumar, 2018-10-17:
MSCKF_VIO is working smoothly with my  i7 GTX1060 pc but performance in TX2 is very poor.
Did anyone try with TX2?

ke-sun, 2018-10-17:
Did you enable jetson clock?

Also, the frame rate (we used 20fps for TX2) and number of features (1~2 features for each grid) should be reduced to adapt the computation power.

You can also use the gpu-accelerated feature detection and tracking functions comes with the OpenCV. However, you have to compile the opencv yourself then since the ros-distro-opencv3 is not compiled with cuda support.

arjunskumar, 2018-10-21:
Yes, I enabled the jetson clock.
I changed the features to 1-2 as well and installed Opencv with Cuda support.
But still not able to process smoothly with bag files as well real camera. The same bag file recorded with jetson working fine with my  i7 GTX1060 pc

kartikmohta, 2018-10-23:
@arjunskumar Just checking, but did you compile with CMAKE_BUILD_TYPE set to Release mode? (See http://wiki.ros.org/catkin/Tutorials/using_a_workspace#With_catkin_make)

arjunskumar, 2018-10-24:
@kartikmohta @ke-sun : Release mode made it works successfully.

closed 2018-10-24 by arjunskumar