KumarRobotics/msckf_vio #58: DoF in gating test

symao, 2018-10-18:
Before measurement update, there is a gating test for outlier rejection, the chi-square test inside gating test is related to the number of degrees of freedom, while there are two DoF formula inside code:
1. in pruneCamStateBuffer(), DoF is involved_cam_state_ids.size()
2. in removeLostFeatures(), DoF is cam_state_ids.size()-1

what is the -1 for? 
Besides, I suppose the dof in chi-square test to be the size of residual, which is 4n-3, where n is camera counts. Why it uses n-1 or n?

ke-sun, 2018-10-19:
I am kind of rusty on the dof of chi-square test. But at the end of day, it is just a threshold for the normalized error. So, the choice are mostly based on experience.

moshanATucsd, 2019-03-19:
Maybe dof in gating test should be camera number + 1, as in ORB SLAM?

ZhouXiner, 2020-03-10:
In the gating test function, may I get any books to understand how the way gamma solved works. Thanks!

moshanATucsd, 2020-03-10:
@ZhouXiner some references 

* [paper](https://pdfs.semanticscholar.org/281b/8f6839e0864fb9f40890bb9b52115edbdc9d.pdf)
 * multiple view geometry book by, Hartley Zisserman page 119

closed 2021-02-02 by ke-sun

ZhouXiner, 2021-02-04:
> @ZhouXiner some references
> 
> * [paper](https://pdfs.semanticscholar.org/281b/8f6839e0864fb9f40890bb9b52115edbdc9d.pdf)
> * multiple view geometry book by, Hartley Zisserman page 119

Thanks