KumarRobotics/msckf_vio #64: two-point ransac model

struggleforbetter, 2018-11-20:


struggleforbetter, 2018-11-20:
about the two-point ransac model,after rescalePoints (),the t maybe change.
suppose:
t:(1,1,1)^T
R:I（Identity matrix）
x2：（1/2,0,1)^T
x1：（0,-1,1)^T
x2^T * [t]x * R *x1 == 0

now suppose scale_factor = 2
x2=>x2' ,x1=>x1' (only change x,y coordinate)
x2':(1,0,1)
x1':(0,-2,1)
now，x2'^T * [t]× * x1' /= 0
I have another question:
suppose p_1,P_2 are the features in previous and current normalized plane .they are satisfied with :
p_2^T * [t]× * R * p_1 = 0;when taking matrix H = (s,0,0;0,s,0;0,0,1) to p_1,p_2,then p_1' = Hp_1,p_2' = Hp_2. does a R' ,t' exist to p_2'^T [t']× * R' * p_1' = 0? that is ,H*[t]× * R *H == [t']× * R' ? 
sorry for my poor English,sorry

ke-sun, 2018-11-20:
* Probably the second question first:

Not sure for a general essential matrix. But for the cases where R is identity, the translation vector corresponds to the scaled points is t'=(tx/s, ty/s, tz/s^2), where t=(tx, ty, tz).

* For the first question:

Yes, you are right. Thanks for the catch. It is a bug in the implemented two-point RANSAC. Thanks to the second problem you pointed out, it seems rescaling the points wont' affect finding the correct inliner set. It is just the fitted model is wrong. But I do agree it requires a proper fix to avoid confusion.

struggleforbetter, 2018-11-21:
Thank you very much.It helps me  a lot.

realdealneil, 2018-12-07:
Since rescaling the points does not affect finding the correct inlier set, do you believe you'd get the same results without scaling at all?  Or, does the scaling help with numerical errors?

closed 2021-02-02 by ke-sun