KumarRobotics/msckf_vio #67: About stateAugmentation()

struggleforbetter, 2018-11-27:
[image]
`
Matrix<double, 6, 21> J = Matrix<double, 6, 21>::Zero();
  J.block<3, 3>(0, 0) = R_i_c;
  J.block<3, 3>(0, 15) = Matrix3d::Identity();
  J.block<3, 3>(3, 0) = skewSymmetric(R_w_i.transpose()*t_c_i);
  //J.block<3, 3>(3, 0) = -R_w_i.transpose()*skewSymmetric(t_c_i);
  J.block<3, 3>(3, 12) = Matrix3d::Identity();
  J.block<3, 3>(3, 18) = Matrix3d::Identity();
`

minxuanjun, 2019-02-22:
@struggleforbetter you can test this code
   J.block<3, 3>(3, 0) = -R_w_i.transpose()*skewSymmetric(t_c_i);
    J.block<3, 3>(3, 12) = Matrix3d::Identity();
    J.block<3, 3>(3, 18) = R_w_i.transpose();

struggleforbetter, 2019-03-21:
> @struggleforbetter you can test this code
> **J.block<3, 3>(3, 0) = -R_w_i.transpose()*skewSymmetric(t_c_i);**
> J.block<3, 3>(3, 12) = Matrix3d::Identity();
> J.block<3, 3>(3, 18) = R_w_i.transpose();

**J.block<3, 3>(3, 0) = R_w_i.transpose()*skewSymmetric(t_c_i);**
I think there is no minus

closed 2021-02-02 by ke-sun