KumarRobotics/msckf_vio #68: the timeoffset of my sensor is about 0.1s from kalibr, should I add the imfluence of timeoffset in the code ?

chenwinki, 2018-11-28:
Dear author:
the timeoffset of my sensor is about 0.1s from kalibr, should I add the imfluence of timeoffset in the code?
I find your work did not consider the timeoffset between imu and cameras, whether the 0.1s will cause a bad result? hope for your reply, Thank you!

yooooki, 2018-11-29:
Based on my knowledge, milliseconds of misalignment of imu time stamp could significantly decrease the accuracy. If your timeoffset is stable, subtract it directly should work( in your case, you may need to maintain a buffer because 0.1s is a big number, where a 20hz image source could miss 2 frames).
If your device does not provide you stable time stamp, or you manually add a time stamp when you receive the message, it could be really hard to cover it. You need a better device to continue, or just give up visual-inertial approach. 
Another situation is that altough your offset is stable in a run, it varies randomly in each run. VINS-MONO takes the offset into the state, you can go and have a try.

chenwinki, 2018-11-29:
@yooooki Thank you very much! you just give me a good answer. could you help me for another question,
when I run msckf, the time used in the featureCallback(const CameraMeasurementConstPtr& msg) fuction Is longer than 1/frame_rate, and the picture show the info when run msckf
[image]

closed 2021-02-02 by ke-sun