KumarRobotics/msckf_vio #70: Quickly lose track on EuRoC Vicon sequences and release mode

peter-zhao, 2018-12-10:
I was trying to reproduce some of the results for msckf on EuRoC sequences, with the scripts provided from this repo.  While the output looks reasonable initially (e.g. the green dots), the odom estimation quickly blows up after the MAV takes off (e.g. the red dots).
[image]
The package is built under release mode, and the image processing front-end is functioning:
[image]

Is there anything I missed? Thank you for any suggestions or hints :)

ke-sun, 2018-12-10:
Could you show some screen snapshots for the terminal while the software is running?

peter-zhao, 2018-12-10:
@ke-sun Sure thing. Thank you for the help!
[image]

PS the full screen log is available via the dropbox link:
https://www.dropbox.com/s/dpma6355vc7h15o/on_screen_log.txt?dl=0

ke-sun, 2018-12-10:
It seems the software is not compiled in release mode if you are using a normal laptop/desktop. Otherwise, the computer is probably slow, you have to dial down the number of features.

peter-zhao, 2018-12-10:
Ah I see what's happening.  I was using a custom build opencv3 for my catkin workspace, which might not be built in Release mode.  Switching back to the kinetic default works.  Thank you for the suggestion!

closed 2018-12-10 by peter-zhao