KumarRobotics/msckf_vio #71: Few features run with MYNT-EYE

GuaguaguaLiu, 2018-12-19:
Hello, thanks for your great job and open source sharing. 
I try to run S-MSCKF with MYNT-EYE (D1000) (I have gotten the device parameters) in real time, but can hardly detect feature points. I don't know why. Look forward to your reply. Thank you!
[image]
[image]
[image]

ke-sun, 2018-12-21:
This happens either because the calibration is not accurate enough or the cameras are not synchronized.

AbyssGaze, 2018-12-26:
Hi @GuaguaguaLiu 
It is very likely that the calibration parameters are inaccurate.
I also used mynteye camera, my calibration parameters are as below:

> %YAML:1.0
cam0:
  T_cam_imu:
    [0.004333484028780699, 0.9998993957798755, -0.013506266516427737, 0.049255621055472265,
    -0.999977905916052, 0.004264951462670086, -0.005098810524933442, 0.006192861529920555,
    -0.005040693991942566, 0.013528063721817071, 0.9998957860177322, -0.014821571698691973,
    0.0, 0.0, 0.0, 1.0]
  camera_model: pinhole
  distortion_coeffs: [-0.304389450911487, 0.08243019622631483, -0.0001842415692688673, -0.0016872307389398313]
  distortion_model: radtan
  intrinsics: [370.3189915369928, 370.08017425881485, 376.812343690345, 231.73306904533828]
  resolution: [752, 480]
  timeshift_cam_imu: 0.0038270640059334653
cam1:
  T_cn_cnm1:
    [0.9999430961265735, -0.0017428721610685386, 0.010524585760503222, -0.12272610545082556,
    0.0017480717201660508, 0.9999983545764745, -0.0004848603949443476, -0.0003773176383777829,
    -0.010523723393417873, 0.0005032305352442027, 0.9999444974622167, 0.000453672405193498,
    0.0, 0.0, 0.0, 1.0]
  camera_model: pinhole
  distortion_coeffs: [-0.3308491755680978, 0.10826874636568513, 0.0003066965101301845, 0.0016795692245386296]
  distortion_model: radtan
  intrinsics: [372.4207433648589, 372.5047406441362, 388.5435353169618, 223.32880019991458]
  resolution: [752, 480]
  timeshift_cam_imu: 0.003841518951950367
T_imu_body:
  [1.0000, 0.0000, 0.0000, 0.0000,
  0.0000, 1.0000, 0.0000, 0.0000,
  0.0000, 0.0000, 1.0000, 0.0000,
  0.0000, 0.0000, 0.0000, 1.0000]
T_cam_gps:
  [1.0000, 0.0000, 0.0000, -0.0800,
  0.0000, 1.0000, 0.0000, -0.3000,
  0.0000, 0.0000, 1.0000, -0.2850,
  0.0000, 0.0000, 0.0000, 1.0000]

closed 2021-02-02 by ke-sun