KumarRobotics/msckf_vio #73: trajectory drifts in flight

Tr-H, 2019-01-03:
Hello,

First, thanks for the open-source code. I run the msckf_vio on a UAV with FLIR FL3-U3-13E4C-C stereo cameras and MicroStrain 3DM-GX5-25 imu. The cameras are synchronized by external trigger at 20Hz. My camera-imu calibration results by kalibr is there
[report-imucam-2018-11-21-11-04-12.pdf](https://github.com/KumarRobotics/msckf_vio/files/2722995/report-imucam-2018-11-21-11-04-12.pdf)
But there are some problems with the setup of the noise of acc and gyro. How should I set it up?
I compared accelerometer data with the fla dataset.
[image]

The result of msckf_vio is here. The red is the groundtruth and the green is trajectory from msckf_vio.
[image]
[image]
[image]
How should I improve my performance?

ke-sun, 2019-01-03:
There are too many features set in the launch file. The update cannot be done in time. Reduce the number of features would help. Normally, around 50 features should be enough.

Tr-H, 2019-01-04:
Thanks for your timely reply. I reduced the number of features to around 50, which are distributed in 50 grid like this
[image]
[image]
The result of msckf_vio is here. The red is the groundtruth and the green is trajectory from msckf_vio.
[image]
[image]
[image]
The UAV only runs within 3 meters, showing a large drift.

The strategy of camera exposure synchronization refers to your paper, Experiments in Fast, Autonomous, GPS-Denied Quadrotor Flight. 

The noise set in the launch file is here
[image]
Looking forward to your reply.

ke-sun, 2019-01-04:
Does the problem still exist if the sensors are moved by hand (without turning on the props)? If the problem disappears, it is probably due to the significant IMU noise caused by the vibration of the props and motors. 

Be sure to start the software before turning on the props, since it reads IMU messages for a few seconds to determine the initial orientation, gravity magnitude, and gyro bias. If the software was started before the motors, probably a weak damper is need for the whole sensor package.

ke-sun, 2019-01-04:
BTW, it is probably beneficial to recalibrate the cameras carefully. The reprojection error is a bit large. Normally, error should be under 1.0 pixel.

Tr-H, 2019-01-07:
Thanks for your timely reply. The problem disappears when the sensors are moved by hand. I Recalibrated the sensors by kalibr and the reprojection error is about 0.8. The noise set is the same. The performance during the flight has been significantly improved. Thank you very much for your answer. 

But now I have another question that the trajectory drifts suddenly when camera's exposure time changing. There is some mutation in image brightness as the exposure time changes.

ke-sun, 2019-01-07:
It is a common problem for stereo images. Our fix in the driver is to have the two cameras have the same exposure time with a customized auto-exposure time controller. See #41 for more details.

Tr-H, 2019-01-08:
Thank you for your kindly help! I will adjust params of the exposure controller to improve the performance. Learn a lot from your reply~ Thanks again

closed 2019-01-08 by Tr-H