KumarRobotics/msckf_vio #74: EKF update confusion

fushi219, 2019-01-10:
Hello, thanks for your great job.
I got a question about  the MsckfVio::measurementUpdate function.
When it comes to update the error state,which has been done in the line-966: 

`  // Compute the error of the state.
  VectorXd delta_x = K * r_thin;`

I'm confused that the calculated delta_x  here seems to be just the correction part  of the error state **X** ,which means that the error state should be "**X**+delta_x" after EKF update. Then I could use the new error state(**X**+delta_x) to correct my nominal state.
If so ,why the code here only use the correction part of error state (delta_x) to do correction of nominal state ?Where is my original error state **X** ?

ke-sun, 2019-01-10:
The mean of the error state is always **0** in the prior distribution. Therefore, whatever ```delta_x``` is, it is the updated mean of the error state, which is then directly used to update the nominal state.

fushi219, 2019-01-10:
I got it，thank you so much！ :-)

closed 2019-01-10 by fushi219