KumarRobotics/msckf_vio #77: Issues with transformation from base to camera

Styxny, 2019-03-08:
Hello,

I am using this State Estimator in conjunction with RotorS simulating a firefly drone. 
I am having an issue calculating the T_cam_imu  (listed as transformation from IMU to camera coordinates) for the left and right cameras.  From ros I am getting:

##################
#Tmatrix

rosrun tf tf_echo firefly/base_link firefly/vi_sensor/vi_sensor_link
- Translation: [0.100, 0.000, -0.030]
- Rotation: in Quaternion [0.000, 0.050, 0.000, 0.999]
            in RPY (radian) [0.000, 0.100, 0.000]
            in RPY (degree) [0.000, 5.730, 0.000]

rosrun tf tf_echo firefly/vi_sensor/vi_sensor_link firefly/vi_sensor/camera_left_link
At time 0.000
- Translation: [0.015, 0.055, 0.006]
- Rotation: in Quaternion [0.000, 0.000, 0.000, 1.000]
            in RPY (radian) [0.000, -0.000, 0.000]
            in RPY (degree) [0.000, -0.000, 0.000]
          
rosrun tf tf_echo firefly/vi_sensor/vi_sensor_link firefly/vi_sensor/camera_right_link
At time 0.000
- Translation: [0.015, -0.055, 0.006]
- Rotation: in Quaternion [0.000, 0.000, 0.000, 1.000]
            in RPY (radian) [0.000, -0.000, 0.000]
            in RPY (degree) [0.000, -0.000, 0.000]

rosrun tf tf_echo firefly/vi_sensor/camera_left_link firefly/vi_sensor/camera_left_optical_link
- Translation: [0.000, 0.000, 0.000]
- Rotation: in Quaternion [-0.500, 0.500, -0.500, 0.500]
            in RPY (radian) [-1.571, -0.000, -1.571]
            in RPY (degree) [-90.000, -0.000, -90.000]

#T IMU matrix

rosrun tf tf_echo firefly/base_link firefly/imu_link
- Translation: [0.000, 0.000, 0.000]
- Rotation: in Quaternion [0.000, 0.000, 0.000, 1.000]
            in RPY (radian) [0.000, -0.000, 0.000]
            in RPY (degree) [0.000, -0.000, 0.000]

######################

Which I translate into:
T_Sensor_Base =
[cos(0.1), 0, -sin(0.1),0.1
     0, 1, 0, 0
sin(0.1), 0, cos(0.1), -0.03
0,0,0,1]

T_CameraL_Sensor=
[1, 0, 0, 0.015
 0, 1, 0, 0.055
 0, 0, 1, 0.0.006
 0, 0, 0, 1]

T_CameraR_Sensor=
[1, 0, 0, 0.015
 0, 1, 0, -0.055
 0, 0, 1, 0.0.006
 0, 0, 0, 1]

So then 
T_CameraL_Base =  dot(T_CameraL_Sensor,T_Sensor_Base)
and 
T_CameraR_Base = dot(T_CameraR_Sensor, T_Sensor_Base)
and 
T_C1_C0 = T_CameraR_CameraL =
[1,0,0,0
 0,1,0,-0.11
 0,0,1,0
 0,0,0,1]

But when I plug this in the state estimator diverges.  Am I constructing my matrices incorrectly?

ke-sun, 2019-03-11:
Probably the following helps in resolve the confusion. The `T_f0_f1` in the config files takes a vector from frame `f1` to frame `f2`.

Styxny, 2019-03-14:
Do you mean T_f0_f1 takes a vector from frame f1 to f0?

ke-sun, 2019-03-14:
Oh, sorry. I mean f0 to f1. It is fixed in the last reply.

Styxny, 2019-03-14:
The explanation on this page seems to say that T_f0_f1 is a transformation from f1 to f0, which, from my understanding is the opposite of your explanation, is this page wrong?

https://github.com/ethz-asl/kalibr/wiki/yaml-formats

_T_cam_imu
IMU extrinsics: transformation from IMU to camera coordinates (T_c_i)_

ke-sun, 2019-03-14:
Yes, you are correct. I fixed my previous comment. Sorry for the confusion.

kartikmohta, 2019-03-14:
@Styxny The T_CameraR_CameraL you posted above seems wrong since it shows a stereo translation along the Y-axis. In case you're not, you should be using the camera_optical_link frames for computing the stereo extrinsics to put in the calib file. I actually don't know why they have the two separate frames associated with each camera.

Styxny, 2019-03-15:
@kartikmohta I am looking at that, but it is my understanding that both [camera right to camera right optical link] and [camera left to camera left optical link] are the same transformation, so I am only getting a translation in the Z direction (I fixed my above post from a translation in Y)

kartikmohta, 2019-03-15:
That still seems wrong, the stereo camera translation should be along the X-axis (for a horizonal stereo).

Also, the T_cam_imu is the transformation that converts from the IMU frame to the camera optical link frame (see https://github.com/ethz-asl/rotors_simulator/wiki/Adding-a-Camera-to-RotorS). I would recommend you to ignore the camera_left_link and camera_right_link frames and only use the *_optical_link frames.

Styxny, 2019-03-18:
@kartikmohta I am now only using *_optical_link_frames, and I have attached my python code that I am using to generate the matrices, if you see an issue let me know, but using the code below the camera translation is along the Z axis....  But my state estimator is still diverging, so something is still wrong....

#########
beta = 0.1

''' Transformation pose of Sensor link relative to IMU '''
imuSensor =np.matrix([[np.cos(beta),0,np.sin(beta),0.1],[0,1,0,0],[-np.sin(beta),0,np.cos(beta),-0.03],[0,0,0,1]])

''' Transformation pose of Camera left link relative to Sensor'''
sensorCamL = np.matrix([[1,0,0,0.015],[0,1,0,0.055],[0,0,1,0.006],[0,0,0,1]])

''' Transformation pose of Camera right link relative to sensor '''
sensorCamR = np.matrix([[1,0,0,0.015],[0,1,0,-0.055],[0,0,1,0.006],[0,0,0,1]])

''' X axis and Z axis transformations to get pose of Camera X Optical link relative to Camera X '''
RX = np.matrix([[1,0,0,0],[0,0,-np.sin(-1.571),0],[0,np.sin(-1.571),0,0],[0,0,0,1]])

RZ = np.matrix([[0,-np.sin(-1.571),0,0],[np.sin(-1.571),0,0,0],[0,0,1,0],[0,0,0,1]])

''' pose of Camera Optical Link Left relative to IMU '''
T_B_COL = np.dot(imuSensor,np.dot(sensorCamL,np.dot(RX,RZ)))

''' pose of IMU relative to Camera Optical Link Left
T_COL_B = np.linalg.inv(T_B_COL)

''' pose of IMU relative to Camera Optical Link Right '''
T_B_COR = np.dot(imuSensor,np.dot(sensorCamR,np.dot(RX,RZ)))

''' pose of IMU relative to Camera Optical Link Right '''
T_COR_B = np.linalg.inv(T_B_COR)

''' Pose of Camera Optical Link Left relative to Camera Optical Link Right '''
T_S_COL = np.dot(sensorCamL,np.dot(RX,RZ))
T_COR_S = np.linalg.inv(np.dot(sensorCamR,np.dot(RX,RZ)))
T_COR_COL = np.dot(T_COR_S,T_S_COL)

print('pose IMU rel to COL \n', T_COL_B)

print('pos IMU rel to COR \n', T_COR_B)

print('pose COL rel to COR \n', T_COR_COL)
#############

kartikmohta, 2019-03-20:
In that code, you should replace all the `np.dot(RX,RZ)` with `np.dot(RZ,RX)`. The usual convention in ROS to go from roll, pitch, yaw angles to rotation matrix is to multiply in the order `Rz(yaw) * Ry(pitch) * Rx(roll)`.

closed 2021-02-02 by ke-sun