KumarRobotics/msckf_vio #78: About the track bias

RxhN, 2019-03-20:
Hi ,Thank you for you sharing.
I have some problem about track.
I use zed and xsen to runing . however,almost all of my track result have a little angle/heading bias. as the picture below.
I am waiting for your help.
[image]
[image]
[image]

ke-sun, 2019-03-20:
It is a common problem for vios. Either large or small, there is drift in yaw and position eventually. Tuning the noise parameters can help. But if drift-free is required, it is probably better to look for slam solutions.

jian-li, 2019-07-23:
First, how well you imu and camera are synchronized, this should have big impact on the result. Second, vio is a odometry in nature, but for your situation that you run the system multiple times at the environment, you can build a map and track the map, this is the method that tango use, you can read the maplab project https://github.com/ethz-asl/maplab.

closed 2021-02-02 by ke-sun