KumarRobotics/msckf_vio #79: Using the Error State to calculate the State Transition matrix

g-spacewhale, 2019-03-22:
In eq (10) of the Mourikis 2017 paper, the matrix F is calculated based on the *error-state* description of the imu state. It is then used in eq. (13) to calculate phi - the *state* transition matrix. I understand, that the error state is used because it has non-quaternion based formulation of rotations. But why can the the resulting matrix then be used to calculate the state transition?

kartikmohta, 2019-03-22:
The state transition matrix computed in eq (13) is for the error state, not the actual state. The filter operates only on error-state.

g-spacewhale, 2019-03-23:
I have now read through the concept of error-state estimation in general in 'Aided Navigation, Jay Farrell' and the error-state kalman filter in particular in '[Quaternion kinematics for the error-state Kalman filter](http://www.iri.upc.edu/people/jsola/JoanSola/objectes/notes/kinematics.pdf). 

Do I understand correctly, that this ESKF formulation is primarily used, because the orientation information in the error state has minimal representation? 

What are the disadvantages of such a filter compared to the standard approach of estimating the nominal state? As I understand it, there is no additional loss of information, when linearizing around the error-state vs. the nominal state, so the result should be equivalent.

Please excuse my naive questions regarding this topic,
best wishes

kartikmohta, 2019-03-29:
As you said, the ESKF is primarily used to keep the correct degrees-of-freedom in the covariance even though the actual state representation is not the minimal representation. The only disadvantage is that it's slightly harder to understand but performance-wise it should be equivalent to a normal EKF if the state is in Euclidean space.

g-spacewhale, 2019-03-29:
Thank you very much for your response,
I have worked myself through the linked literature and the process is now clear to me.

closed 2019-03-29 by g-spacewhale