KumarRobotics/msckf_vio #80: Odometry drifts off for stationary pose

Jwenda, 2019-03-25:
Dear community,
I'm facing a problem while running MSCKF in the Gazebo simulation. We have a Realsense D435 camera in the simulator running. I made a record of the running simulator and attached it to the post. The MSCKF is initially started from the static position on the ground, but the result is as same as started in the air.

I also attached the calibration file we used.
[out-2.zip](https://github.com/KumarRobotics/msckf_vio/files/3003391/out-2.zip)
[realsense_simulation.txt](https://github.com/KumarRobotics/msckf_vio/files/3003390/realsense_simulation.txt)

Do I miss something in the configuration file?

Thank you for your help!

Best,
Jan Bednar

poaongithub, 2019-04-08:
same as mine, have you solved it? @Jwenda

Jwenda, 2019-04-08:
No, I'm trying different approaches, but the result is still the same. It drifts off a lot.

jian-li, 2019-07-23:
The msckf algorithm drift when it is hovering, for the reason you can read this paper:https://www-users.cs.umn.edu/~stergios/papers/IROS_2013_Hover_Dimitris_Kejian.pdf. To avoid this, a special strategy should be token or just combine ekf-slam and msckf.

wntun, 2020-08-03:
When I reduce the threshold values for rotation and translation, increase tracking rate threshold, results got improved. Hope it's helpful.

closed 2021-02-02 by ke-sun