KumarRobotics/msckf_vio #83: Two definition of translation treshold values

Jwenda, 2019-06-06:
Dear developers,
I got through the last commit and I realized, you define in the msckf_vio*.launch file _rotation_threshold_ and _feature/config/translation_threshold_. The _rotation_threshold_ is now properly passed, but the _feature/config/translation_threshold_ is just saved but not used at all. What is the purpose of this parameter?

Thanks for the explanation!
Jan

ke-sun, 2019-06-06:
The parameter `feature/config/translation_threshold` is used at the following location. It is used to decide whether to triangulate the 3-D position of a feature based on the translation between the camera poses. Currently, it is set to `-1` meaning that a feature is always triangulated when it is needed.

https://github.com/KumarRobotics/msckf_vio/blob/0fc32805fc7b1edb0b7db24dcd594b3464c8e2d9/include/msckf_vio/feature.hpp#L287-L289

Jwenda, 2019-06-06:
Ohh, okay, I missed that. Thank you very much!

closed 2019-06-06 by Jwenda