KumarRobotics/msckf_vio #85: Huber robust cost

moshanATucsd, 2019-07-24:
Hi, for Huber robust cost in [void Feature::jacobian](https://github.com/KumarRobotics/msckf_vio/blob/a9386c5d2bf609263a928108df6453b68672bc24/include/msckf_vio/feature.hpp), I am wondering how to get the following

```
  // Compute the weight based on the residual.
  double e = r.norm();
  if (e <= optimization_config.huber_epsilon)
    w = 1.0;
  else
    w = optimization_config.huber_epsilon / (2*e);
```

shouldn't this be 

```
  double e = r.norm();
  if (e <= optimization_config.huber_epsilon)
    w = 1.0;
  else
    w = np.sqrt( optimization_config.huber_epsilon / (e));
```

reference
TABLE 2 in [An Evaluation of Robust Cost Functions for RGB Direct Mapping](https://www.researchgate.net/publication/280227031_An_Evaluation_of_Robust_Cost_Functions_for_RGB_Direct_Mapping)

ke-sun, 2019-07-25:
Nice catch! I did make a poop here :smile: 

Something more to double check before fixing this:

The Huber norm is e'=2de - d^2 for large e (e is the error, d is the epsilon in Huber norm). It's a bit different from wikipedia since we don't have 1/2 for the errors.  Since d^2 does not involve the variables, the Huber norm here can be safely treated as e'=2de.

Because the code here (how we use the weight and error later):
https://github.com/KumarRobotics/msckf_vio/blob/0fc32805fc7b1edb0b7db24dcd594b3464c8e2d9/include/msckf_vio/feature.hpp#L373-L375

the weight should be set as w=sqrt(2de)/e = sqrt(2d/e). What do you think?

moshanATucsd, 2019-07-25:
Hi, thanks for the reply! I think you are right. I just found the Huber norm from [dso](https://github.com/JakobEngel/dso/blob/e59db70710ead30d6b47959f886665ddd9cb4f85/src/FullSystem/Residuals.cpp)

```
		float hw = fabsf(residual) < setting_huberTH ? 1 : setting_huberTH / fabsf(residual);
		energyLeft += w*w*hw *residual*residual*(2-hw);

		{
			if(hw < 1) hw = sqrtf(hw);
```

looks the same as what you said

JzHuai0108, 2019-11-26:
Do we need 2.0 in [line](https://github.com/KumarRobotics/msckf_vio/blob/master/include/msckf_vio/feature.hpp#L223)?
My derivation agrees with the suggestion of @moshanATucsd on Jul 24. 
Please also refer to the ceres solver for computing the [Huber loss first order derivative](https://github.com/ceres-solver/ceres-solver/blob/94712db5c70116f1832fbedbfbf98f26923e8b48/internal/ceres/loss_function.cc#L54) which is used to rescale the [residuals](https://github.com/ceres-solver/ceres-solver/blob/189f0c2f6efd07c3976953a5c278dbb5f263183b/internal/ceres/corrector.cc#L82) and [Jacobians](https://github.com/ceres-solver/ceres-solver/blob/189f0c2f6efd07c3976953a5c278dbb5f263183b/internal/ceres/corrector.cc#L138).

moshanATucsd, 2019-11-26:
Hi @JzHuai0108, I think I agree with you, even if we do not have 1/2 for the errors, we still don't need the 2, eg refer to eq. 50 in [paper](http://ethaneade.com/optimization.pdf). What do you think? @ke-sun

closed 2021-02-02 by ke-sun