KumarRobotics/msckf_vio #86: Using our own sensor

moshanATucsd, 2019-10-10:
Hi, I am wondering if we want to run this algorithm using our own sensor, is there any restrictions? Eg what's the range of IMU frequency, image freq. etc. should work.

ke-sun, 2019-10-12:
1. The `frame_rate` parameter is not important. It is used nowhere but to check the processing time of the filter is within the frame interval.
2. The convention of the transformation in the calib file agrees with the kalibr. `T_x_y` represents a transformation that takes a vector from frame `y` into frame `x`.

For the warning `Images seems unsynced`, usually there are two causes. Either the calibration is not accurate, or the stereo images are really not synchronized. Both can cause failure in searching for matches between stereo images.

A easy way to tell which one is the actual cause is to make the sense sits statically and check if the warning goes away. If you find the warning disappears and there are expected number of features matched, it means the stereo images are not synchronized. If the warning remains, there is probably still something wrong in the calibration.

Here is the block that triggers the warning:
https://github.com/KumarRobotics/msckf_vio/blob/e3a39a95b57da376924aa22b46637b60f68f504c/src/image_processor.cpp#L773-L777

moshanATucsd, 2019-10-12:
Thank you so much @ke-sun, really appreciate your quick reply. 

You said `T_x_y represents a transformation that takes a vector from frame y into frame x`, which seems different from`camx/T_cam_imu: takes a vector from the IMU frame to the camx frame. cam1/T_cn_cnm1: takes a vector from the cam0 frame to the cam1 frame.` in readme.  I am still confused whether T_cn_cnm1 is from cam0 to cam1. 

I have held the sensor static and the warning remains, a screenshot is below.  It's wired that no features are tracked initially, and later on few features are detected. I am wondering whether there are feature detection related configurations in the launch file that we have to change? 

[image]

Here's the output when sensor is static
```
[ INFO] [1570914437.984274641]:  candidates: 30; track: 30; match: 26; ransac: 19/30=0.633333
[ WARN] [1570914437.991376597]: Images at [1570570727.560277] seems unsynced...
[ WARN] [1570914438.023179944]: Images at [1570570727.580649] seems unsynced...
[ WARN] [1570914438.051043106]: Images at [1570570727.600033] seems unsynced...
[ WARN] [1570914438.077951941]: Images at [1570570727.619849] seems unsynced...
[ WARN] [1570914438.129096211]: Images at [1570570727.639983] seems unsynced...
[ WARN] [1570914438.161022640]: Images at [1570570727.659805] seems unsynced...
[ WARN] [1570914438.221840600]: Images at [1570570727.699836] seems unsynced...
[ WARN] [1570914438.248324729]: Images at [1570570727.719825] seems unsynced...
[ WARN] [1570914438.322583774]: Images at [1570570727.779656] seems unsynced...
[ WARN] [1570914438.348175367]: Images at [1570570727.800142] seems unsynced...
[ WARN] [1570914438.428054377]: Images at [1570570727.839964] seems unsynced...
[ WARN] [1570914438.478846107]: Images at [1570570727.879616] seems unsynced...
[ INFO] [1570914438.507620805]:  candidates: 6; track: 5; match: 0; ransac: 0/6=0.000000
[ WARN] [1570914438.548395763]: Images at [1570570727.959537] seems unsynced...
[ WARN] [1570914438.579353740]: Images at [1570570727.999382] seems unsynced...
[ WARN] [1570914438.607023237]: Images at [1570570728.039310] seems unsynced...
[ WARN] [1570914438.634870649]: Images at [1570570728.059135] seems unsynced...
[ WARN] [1570914438.691556302]: Images at [1570570728.118747] seems unsynced...
[ WARN] [1570914438.716195942]: Images at [1570570728.138889] seems unsynced...
[ WARN] [1570914438.772894886]: Images at [1570570728.198814] seems unsynced...
[ WARN] [1570914438.803208366]: Images at [1570570728.219654] seems unsynced...
[ WARN] [1570914438.871781737]: Images at [1570570728.278806] seems unsynced...
[ WARN] [1570914438.903263628]: Images at [1570570728.318821] seems unsynced...
[ WARN] [1570914438.941052604]: Images at [1570570728.358698] seems unsynced...
[ WARN] [1570914438.997244884]: Images at [1570570728.418477] seems unsynced...
[ INFO] [1570914439.025308337]:  candidates: 4; track: 4; match: 4; ransac: 4/4=0.999998
[ WARN] [1570914439.033712369]: Images at [1570570728.458432] seems unsynced...
[ WARN] [1570914439.064781262]: Images at [1570570728.478452] seems unsynced...
[ WARN] [1570914439.124929851]: Images at [1570570728.558506] seems unsynced...
[ WARN] [1570914439.279413236]: Images at [1570570728.697753] seems unsynced...
[ WARN] [1570914439.311416665]: Images at [1570570728.737670] seems unsynced...
[ WARN] [1570914439.341576147]: Images at [1570570728.757653] seems unsynced...
[ WARN] [1570914439.395237068]: Images at [1570570728.797432] seems unsynced...
[ WARN] [1570914439.524187818]: Images at [1570570728.937180] seems unsynced...
[ INFO] [1570914439.551073861]:  candidates: 9; track: 9; match: 9; ransac: 6/9=0.666666
```

ke-sun, 2019-10-12:
If this is the case, the calibration is wrong or not accurate enough. I suggest to use Kalibr or [multicam_calibration](https://github.com/KumarRobotics/multicam_calibration). These are what we use to calibrate the stereo cameras. The output convention is compatible with the requirement of msckf_vio.

moshanATucsd, 2019-10-12:
I see, could you kindly confirm that `cam1/T_cn_cnm1: takes a vector from the cam0 frame to the cam1 frame` in the readme is correct, and T_cn_cnm1 is not from cam1 to cam0?

ke-sun, 2019-10-12:
Yes. What is in the readme is correct.

moshanATucsd, 2019-10-12:
Thank you so much!

closed 2019-10-12 by moshanATucsd