KumarRobotics/msckf_vio #87: No result from MSCKF_VIO

tekrajchhetri, 2019-11-18:
Based on read me in order to run vio the input required input topics are camera and imu input. I have stereo camera input and imu (shown below -- topic renamed). As suggested the robot starts from stationary state. But when I run I get nothing, it runs without error but unable to produce odometry or publish point cloud. Is there anything missing? 
[image]

Thank you.

closed 2019-11-26 by tekrajchhetri

kamzero, 2021-04-19:
I have the same question with you. Could you please tell me how did you solve it?

tekrajchhetri, 2021-04-20:
@kamzero I do not remember completely but it if I am not mistaken the topics needs to be synchronised.

kamzero, 2021-04-21:
Do you mean that the frequency of the same type of rostopic in different config files should be set to the same?