KumarRobotics/msckf_vio #88: Image processing node dies

Jwenda, 2019-11-19:
Dear @ke-sun,
I am trying to tune the algorithm and I have faced an interesting issue. While setting the `grid_col: 6`, the image processing node fails at the second or third iteration:
[image]

I have tried different grid settings regarding number of columns, rows, and features. It always works except using `grid_col: 6`. We are using the 640x360 resolution. I assume it is related to the resolution used, but I haven't properly checked other resolutions.

Could you please check it?

Thanks,
Jan

ke-sun, 2019-11-19:
That's a nice catch. Thanks a lot for pointing that out!

My guess is that 640 is not a multiple of 6. Therefore, there is small strip of images left out. Although new features can be detected over this small strip, they should not be used since they don't fit in any cells in the grid.

Related line are attached below:
https://github.com/KumarRobotics/msckf_vio/blob/e3a39a95b57da376924aa22b46637b60f68f504c/src/image_processor.cpp#L354-L355
https://github.com/KumarRobotics/msckf_vio/blob/e3a39a95b57da376924aa22b46637b60f68f504c/src/image_processor.cpp#L732-L733
New features should be ignored if the `row` and `col` of the new feature is not in range. Something as the following:
```
if (row>=grid_height || col>=grid_width) continue;
```

I will test this later. But just want to let you know before that in case you need some urgent fix.

Jwenda, 2019-11-20:
I have just tested several options, hence I do not require the fix now, but thanks!

I have tried the `col` parameter to 3,5,6 and 7, just the 6 does not work. The division by 3,6 and 7 gives a floating number, but only 6 gives a floating number higher than 0.5, specifically 106.667. Hence I tried `col=14` which gives 45.714 and it fails again, so there seems to be a problem with rounding.

closed 2021-02-02 by ke-sun