KumarRobotics/msckf_vio #93: how about fisheye?

wycjl, 2020-06-01:
thanks for your project, i want to use realsense t265 to run msckf_vio, is there any problem about it?

wang-yu-yang, 2020-07-05:
@wycjl   Did you test successfully with T265? I have encountered some problems. When I modify the launch file as required, the terminal keeps displaying the following content, but there is no any map point in rviz：

SUMMARY
========

PARAMETERS
 * /firefly_sbx/image_processor/T_imu_body: [1.0, 0.0, 0.0, 0...
 * /firefly_sbx/image_processor/cam0/T_cam_imu: [-0.9999001288978...
 * /firefly_sbx/image_processor/cam0/camera_model: pinhole
 * /firefly_sbx/image_processor/cam0/distortion_coeffs: [0.00091421585857...
 * /firefly_sbx/image_processor/cam0/distortion_model: equidistant
 * /firefly_sbx/image_processor/cam0/intrinsics: [286.951132850664...
 * /firefly_sbx/image_processor/cam0/resolution: [848, 800]
 * /firefly_sbx/image_processor/cam0/timeshift_cam_imu: 0.00117503939545
 * /firefly_sbx/image_processor/cam1/T_cam_imu: [-0.9999046743355...
 * /firefly_sbx/image_processor/cam1/T_cn_cnm1: [0.99999885603982...
 * /firefly_sbx/image_processor/cam1/camera_model: pinhole
 * /firefly_sbx/image_processor/cam1/distortion_coeffs: [0.00652443764918...
 * /firefly_sbx/image_processor/cam1/distortion_model: equidistant
 * /firefly_sbx/image_processor/cam1/intrinsics: [286.865305592168...
 * /firefly_sbx/image_processor/cam1/resolution: [848, 800]
 * /firefly_sbx/image_processor/cam1/timeshift_cam_imu: -0.00339846837944
 * /firefly_sbx/image_processor/fast_threshold: 10
 * /firefly_sbx/image_processor/grid_col: 5
 * /firefly_sbx/image_processor/grid_max_feature_num: 4
 * /firefly_sbx/image_processor/grid_min_feature_num: 3
 * /firefly_sbx/image_processor/grid_row: 4
 * /firefly_sbx/image_processor/max_iteration: 30
 * /firefly_sbx/image_processor/patch_size: 15
 * /firefly_sbx/image_processor/pyramid_levels: 3
 * /firefly_sbx/image_processor/ransac_threshold: 3
 * /firefly_sbx/image_processor/stereo_threshold: 5
 * /firefly_sbx/image_processor/track_precision: 0.01
 * /firefly_sbx/vio/T_imu_body: [1.0, 0.0, 0.0, 0...
 * /firefly_sbx/vio/cam0/T_cam_imu: [-0.9999001288978...
 * /firefly_sbx/vio/cam0/camera_model: pinhole
 * /firefly_sbx/vio/cam0/distortion_coeffs: [0.00091421585857...
 * /firefly_sbx/vio/cam0/distortion_model: equidistant
 * /firefly_sbx/vio/cam0/intrinsics: [286.951132850664...
 * /firefly_sbx/vio/cam0/resolution: [848, 800]
 * /firefly_sbx/vio/cam0/timeshift_cam_imu: 0.00117503939545
 * /firefly_sbx/vio/cam1/T_cam_imu: [-0.9999046743355...
 * /firefly_sbx/vio/cam1/T_cn_cnm1: [0.99999885603982...
 * /firefly_sbx/vio/cam1/camera_model: pinhole
 * /firefly_sbx/vio/cam1/distortion_coeffs: [0.00652443764918...
 * /firefly_sbx/vio/cam1/distortion_model: equidistant
 * /firefly_sbx/vio/cam1/intrinsics: [286.865305592168...
 * /firefly_sbx/vio/cam1/resolution: [848, 800]
 * /firefly_sbx/vio/cam1/timeshift_cam_imu: -0.00339846837944
 * /firefly_sbx/vio/child_frame_id: odom
 * /firefly_sbx/vio/feature/config/translation_threshold: -1.0
 * /firefly_sbx/vio/fixed_frame_id: world
 * /firefly_sbx/vio/frame_rate: 20
 * /firefly_sbx/vio/initial_covariance/acc_bias: 0.01
 * /firefly_sbx/vio/initial_covariance/extrinsic_rotation_cov: 0.00030462
 * /firefly_sbx/vio/initial_covariance/extrinsic_translation_cov: 2.5e-05
 * /firefly_sbx/vio/initial_covariance/gyro_bias: 0.01
 * /firefly_sbx/vio/initial_covariance/velocity: 0.25
 * /firefly_sbx/vio/initial_state/velocity/x: 0.0
 * /firefly_sbx/vio/initial_state/velocity/y: 0.0
 * /firefly_sbx/vio/initial_state/velocity/z: 0.0
 * /firefly_sbx/vio/max_cam_state_size: 20
 * /firefly_sbx/vio/noise/acc: 0.05
 * /firefly_sbx/vio/noise/acc_bias: 0.01
 * /firefly_sbx/vio/noise/feature: 0.035
 * /firefly_sbx/vio/noise/gyro: 0.005
 * /firefly_sbx/vio/noise/gyro_bias: 0.001
 * /firefly_sbx/vio/position_std_threshold: 8.0
 * /firefly_sbx/vio/publish_tf: True
 * /firefly_sbx/vio/rotation_threshold: 0.2618
[... 46 lines cut ...]
[ INFO] [1593920605.817938348]: frame rate: 20.000000
[ INFO] [1593920605.817991438]: position std threshold: 8.000000
[ INFO] [1593920605.818045571]: Keyframe rotation threshold: 0.261800
[ INFO] [1593920605.818098282]: Keyframe translation threshold: 0.400000
[ INFO] [1593920605.818149163]: Keyframe tracking rate threshold: 0.500000
[ INFO] [1593920605.819648173]: gyro noise: 0.0000250000
[ INFO] [1593920605.819754690]: gyro bias noise: 0.0000010000
[ INFO] [1593920605.819975752]: acc noise: 0.0025000000
[ INFO] [1593920605.820024866]: acc bias noise: 0.0001000000
[ INFO] [1593920605.820072334]: observation noise: 0.0012250000
[ INFO] [1593920605.820119199]: initial velocity: 0.000000, 0.000000, 0.000000
[ INFO] [1593920605.820174183]: initial gyro bias cov: 0.010000
[ INFO] [1593920605.820227606]: initial acc bias cov: 0.010000
[ INFO] [1593920605.820276729]: initial velocity cov: 0.250000
[ INFO] [1593920605.821297073]: initial extrinsic rotation cov: 0.000305
[ INFO] [1593920605.821372676]: initial extrinsic translation cov: 0.000025
   -0.9999 0.00551683  0.0130114
-0.0054273  -0.999961 0.00690579
  0.013049 0.00683448   0.999892
0.000714929 0.000216749 -0.00384451
[ INFO] [1593920605.821576999]: max camera state #: 20
[ INFO] [1593920605.821644786]: ===========================================
[ INFO] [1593920605.821698848]: Finish loading ROS parameters...
[ INFO] [1593920605.859261025]: Finish creating ROS IO...
[ INFO] [1593920605.890252989]: Finish creating ROS IO...

ke-sun, 2020-07-06:
Probably, topics are not remapped correctly.

wang-yu-yang, 2020-07-06:
Thanks for your help. I will find a way to solve the problem of image correction.

wang-yu-yang, 2020-07-09:
@ke-sun ，I have confirmed that there is no error in the topic. Does msckf_vio support fisheye cameras? I use Intel realsense T265, but the previous problem still appears.

closed 2021-02-02 by ke-sun