KumarRobotics/msckf_vio #100: The Groundtruth

jingshaojing, 2021-01-16:
For fast flight dataset , how can I get the position and velocity of gps?  It is workable for  using  rosbag play?  And i have see the matlab script for dataset, it only use 2*N velocity? what is about  position?

ke-sun, 2021-01-16:
The GPS data are in the rosbag files.
The matlab script is to compute the time and yaw offset between GPS and the odometry.
Once we have the offset, we could transform either the odometry data or the GPS data for proper comparison.

jingshaojing, 2021-01-17:
thanks，  I have understand this meaning , I will try

jingshaojing, 2021-01-19:
I have a new question,for  V1_01_easy.bag，the groundtruth in mav0 has 28713 raws in data.csv, and I save pose data  in txt file only has 2893 raws, My question is whether too little Pose data is being saved？ @ @ke-sun

jingshaojing, 2021-01-31:
Apologize for my disturbance, can i directly use matlab script with flight dataset, I found .m file is only a  function, so  I need to read rosbag file and store relative information in .mat file, and write a main function in matlab to use the function you give ? I do not know my understanding is right?

ke-sun, 2021-02-01:
There seems to be rosbag support in matlab. 
https://www.mathworks.com/help/ros/ref/rosbag.html

jingshaojing, 2021-02-02:
Thanks @ke-sun

closed 2021-02-05 by ke-sun