KumarRobotics/msckf_vio #101: How to debug msckf?

jingshaojing, 2021-01-16:
I want to know how to debug this project in vscode or clion ,except for cmake ........=DEBUG? In other words, I want to study How does this project work out step by step？thanks。

jingshaojing, 2021-01-16:
When I load this project in Clion without "cmake&make",it will hint 
 CMake Error at CMakeLists.txt:8 (find_package):
  By not providing "Findcatkin.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "catkin", but
  CMake did not find one.
when  I change to VScode, it reminds that detects header file inclusion errors.

So how can i debug this?I guess I had to cmake with DEBUG ,then reload this project?

ke-sun, 2021-01-16:
My apologies first. I am not sure about the question.
But the package is supposed to be complied/used in a ROS workspace.
I assume some additional configurations is needed for Clion or VSCode to properly set up the environment.

versatran01, 2021-01-17:
You need to run Clion from command line after sourcing the catkin workspace.
https://www.jetbrains.com/help/clion/ros-setup-tutorial.html

jingshaojing, 2021-01-17:
> You need to run Clion from command line after sourcing the catkin workspace.
> https://www.jetbrains.com/help/clion/ros-setup-tutorial.html

> You need to run Clion from command line after sourcing the catkin workspace.
> https://www.jetbrains.com/help/clion/ros-setup-tutorial.html

I will try this,thanks

jingshaojing, 2021-01-17:
> You need to run Clion from command line after sourcing the catkin workspace.
> https://www.jetbrains.com/help/clion/ros-setup-tutorial.html

 I will try this, thanks. @versatran01

jingshaojing, 2021-01-17:
> My apologies first. I am not sure about the question.
> But the package is supposed to be complied/used in a ROS workspace.
> I assume some additional configurations is needed for Clion or VSCode to properly set up the environment.
> 
> @versatran01 might be an expert on this?

First of all, thank you for providing answers. The original intention is that I am more interested in this kind of project if  you should have relied on a similar IDE (such as Clion or VScode ,though the later is a editor)when you wrote it? So I would like to ask how you wrote this project step by step, or you debugged it in the terminal. 
? 
I will try these two methods(Clion and vscode), maybe I can directly debug in terminal? I hope I can ask if you have any questions @ke-sun

ke-sun, 2021-01-17:
We mostly use "terminal", probably for its "simplicity". :smile:

jingshaojing, 2021-01-17:
so  it means   you are "cmake" project with debug model ,then observe value in  debug model with terminal ？  And finally you get conclusion that if we want to get a good result ，we have to cmake with release  ？Can I understand like this ？sorry for my poor english.I can try with terminal to debug . thanks！

closed 2021-02-05 by ke-sun