KumarRobotics/msckf_vio #106: Doubt about cam1_points initialization

rocomanmelli, 2021-02-03:
Hi!
I have been seeing the code and got a doubt about this [if](https://github.com/KumarRobotics/msckf_vio/blob/e3a39a95b57da376924aa22b46637b60f68f504c/src/image_processor.cpp#L616). As far as I understand, trying to initialize the features of the right camera in that way would only make sense if the 3D points belonged to the same plane (in space) which should not be the case in general. Is it just an approximated guess or I am missing something?
Thanks in advance.

ke-sun, 2021-02-03:
Not sure about the "the same plane" way of thinking. Probably it's easier to think in the following way.

What's in the condition is a way to general initial guesses for matches (which will be later refined through tracking).
The guess is generated by simply projecting the pixel coordinates from the left frame to the right frame.
If the rotation between left and right cameras is identity, and the calibrations parameters of both cameras are the same, `cam1_points` would be just a copy `cam0_points`.
It may not be the best way to generate initial guess, but should be sufficient most times.

https://github.com/KumarRobotics/msckf_vio/blob/e3a39a95b57da376924aa22b46637b60f68f504c/src/image_processor.cpp#L616-L626

rocomanmelli, 2021-02-03:
Thank you for the answer. I understand. When I asked I was thinking about a case like this:
[image]
For P, it would not be a good initial guess, right?

rocomanmelli, 2021-02-03:
Anyway, the condition is not related to belonging to the same plane in space, actually. I was wrong about that.

ke-sun, 2021-02-04:
I see. In that sense, the above initialization assumes all 3-d points are infinitely far away from the image plane.

closed 2021-02-04 by rocomanmelli