KumarRobotics/msckf_vio #107: Question about Feature::generateInitialGuess

TWCV2020, 2021-02-27:
To whom it may concern, I have a question about lines 235 and 236 in Feature::generateInitialGuess in msckf_vio/feature.hpp
It seems that, instead of
A(0) = m(0) - z2(0)*m(2);
A(1) = m(1) - z2(1)*m(2);
we should have
A(0) = m(0) - ( T_c1_c2.translation()(0) / T_c1_c2.translation()(2) ) * m(2);
A(1) = m(1) - ( T_c1_c2.translation()(1) / T_c1_c2.translation()(2) ) * m(2);
The above is obtained by linearizing z2 with respect to depth using first-order Taylor expansion around depth = 0. Please let me know if my understanding is correct. Much appreciated!

ke-sun, 2021-02-27:
I am not sure if an linearization is necessary in this case.

If I remember correctly, the initialize guess is generated by solving D in `Z * z2 = R * D * z1 + t` (D and Z are the depths of the feature in two frames).
The last row of the equation gives `Z = D * m(2) + t(2)` where `m = R * z1`.
Using Z in the first two rows of equations should agree with the code.

TWCV2020, 2021-02-27:
Got it. I was directly linearizing the right hand side of 
z2(0) = (D * m(0) + t(0))/(D * m(2) + t(2))
z2(1) = (D * m(1) + t(1))/(D * m(2) + t(2)),
but we can directly multiply both sides with (D * m(2) + t(2)) as you explained.
Thank you for the clarification!

closed 2021-02-27 by ke-sun