KumarRobotics/msckf_vio #113: Ask for help about evaluation

jingshaojing, 2021-04-21:
I run the work with V1_01_EASY.bag in realse mode , Since I have to use the truth value of the pose matrix, but I have cost too much time for this without result. I will explain my steps as follows , hope someone can answer for me, sorry to bother everyone. 
**step1** : I have V1_01.ZIP，so I get truth quaternion from  vicon folder , and  the posture calculated according to the quaternion is as follows（Because msckf runs V1, this data set only outputs 2893 poses， so I save corresponding 2893 data in data.csv of vicon folder according timestamps)

[image]

only some values , the order is [yaw,pitch ,roll] ,the code is `quat2angle(q)` in matlab code

**step2** I used the value at the first moment in vicon to modify the initial value in the program in _imu_state.h_
`orientation(Eigen::Vector4d(0.0133, -0.0248, -0.1133, 0.9932)),
position(Eigen::Vector3d(0.8564,2.1499,0.9391)),`

Here I considered converting the truth value of vicon (s to R) to R relative to S , corresponding to program.

**step3 :** I get results of msckf , of course , the quaternion is converted to the same corresponding relationship as the truth value （W到I), but the result is
[image]

 It is totally wrong ,and the traj is also as follows 

[image]

though I can get a good  result for traj with evo ,but  now for some reason , I wang to use matlab to deal ,and neither evo or matlab ,the attitude is so difference .

Can some one tell me how to do ?  Only for  a right compare with attitude.

chutsu, 2021-04-27:
To save you time you might want to consider using the Trajectory Evaluation tool by Zichao Zhang:
- Github repo: https://github.com/uzh-rpg/rpg_trajectory_evaluation
- Paper: [Zichao Zhang, Davide Scaramuzza: A Tutorial on Quantitative Trajectory Evaluation for Visual(-Inertial) Odometry, IEEE/RSJ Int. Conf. Intell. Robot. Syst. (IROS), 2018. ](http://rpg.ifi.uzh.ch/docs/IROS18_Zhang.pdf)

In the tool it does all the vicon pose vs MSCKF pose interpolation for you, then it aligns the trajectory as much as it can and calculate the pose error between the vicon and estimated. The tool is pretty well documented, if you know ROS you shouldn't have much trouble using it. Hope this helps

jingshaojing, 2021-04-27:
I would like to try this . yeah ,I know ros and have used EVO to do as above. Hoping for a good result. thanks! @chutsu

closed 2021-04-28 by ke-sun