KumarRobotics/msckf_vio #115: Question about Jacobian of state augmentation?

FanaticsKang, 2021-04-23:
I have a question about your Jacobian in `MsckfVio::stateAugmentation`.
The derivative of *new camera pose* with repect to *the translation bewtween camera and body* seems more accurate as the following: 
```C++
J.block<3, 3>(3, 18) = R_w_i.transpose();
```
Looking forward to your rely.

ke-sun, 2021-04-23:
I think you are right. Thanks for noticing.
Did you have a chance to try the corrected Jacobian? Just wondering if it helps in improving the estimation accuracy.

FanaticsKang, 2021-04-26:
I can't find any difference by EuRoC. 
I think the default covariance is too small or the calibration error is limited.
If msckf is used in our project, I will let u know its performence.

closed 2021-04-26 by FanaticsKang