KumarRobotics/msckf_vio #117: How to improve accuracy?

jingshaojing, 2021-06-28:
First , Thank you very much for your work. It is really a good work . I use D435i for this project. In my view, it may seems work, it has achieved a certain effect in my opinion, but when comparing GPS and RTK, the accuracy is not very satisfactory.
I have tried to modify the following parameters to achieve the current effect:
`"noise/gyro" `   `"noise/acc"`   ` "noise/gyro_bias"`  ` "noise/acc_bias"`  ` "initial_covariance/gyro_bias" `   `"initial_covariance/acc_bias"`  ` "initial_covariance/extrinsic_rotation_cov" `  ` "initial_covariance/extrinsic_translation_cov" `
`"grid_min_feature_num" `   `"grid_max_feature_num" `  ` "max_iteration"`    `"ransac_threshold"`

The following three are my own test videos after referring to all the parameters that can be modified in the issue:
[image]
[image]
[image]

_**So what parameters can I adjust to get better results？**_  Thank you.

ke-sun, 2021-06-29:
That's a tricky question for which I am afraid I don't have a good answer. It depends on lots of factors like testing environments, motion of the sensor, and properties of the actual hardware, etc.

Just based on the attached images, it seems features are tracked and matched fine. The estimated orientation are mostly correct. It's the translation that drifts more than expected. I would try to tune the acceleration noise together with the feature noise to see which one should be relied on more in order to reduce translation error.

With that being said, it is expected for visual inertial odometry to have large error using sensors like D435i given its short stereo baseline and low cost IMU. In case much more accurate estimation is required, loop closure might be necessary.

jingshaojing, 2021-06-29:
Thanks for your advice!

closed 2021-06-29 by jingshaojing