KumarRobotics/msckf_vio #120: How can I fix the position in the window？

wangyuanbiubiubiu, 2021-08-06:
hi，I don't want to adjust the position of the first few frames of the window. What should I do? When I turn down the covariance matrix, there will be a negative diagonal

ke-sun, 2021-08-06:
I am afraid I don't understand the question. Could you elaborate on that?

wangyuanbiubiubiu, 2021-08-07:
I'm sorry I didn't make it clear.
For example, when frame i and frame j co-view the same landmark, it will be adjusted simultaneously when this point feature reprojection error measurement is updated. But I want to hold the frame position, like ”SetParameterBlockConstant“ in the use of Ceres. What should I do in the filtering method?

ke-sun, 2021-08-07:
That would be tricky, since the software is not designed for this purpose.
The first way that pops off my head is to manually set the entries in the measurement Jacobian matrix, that corresponds to the camera frame you would like to fix, to 0.

wangyuanbiubiubiu, 2021-08-07:
In fact, Jacobi has already generated zero, but I still can't fix the position. So I think, in terms of the fixed number of updates, it's probably not the same way as optimization

ke-sun, 2021-08-08:
The problem you mentioned can be related to the prior covariance. If the camera pose to be fixed is correlated with others (camera or IMU states) left to be optimized, the camera pose can still be updated.

wangyuanbiubiubiu, 2021-08-09:
thanks！I found that the window had an overall drift after the reprojection error update, and I wanted to fix this by fixing the first frame pose in the clones_IMU.
And I found another problem, I am running MSCKF in the car. When the car goes straight, roll and pitch are not observable, but when the car turns, roll and pitch are observable. How should I distinguish the two states?

ke-sun, 2021-08-09:
It might be a hacky fix since the algorithm relies on the measurement update to correct the drift in the first place. The drift that you mentioned might be related to the quality of features, calibrations, etc.

It's probably due to the unobservability of gyroscrope bias when there is no rotation in the motion of sensors. Especially for cars, it can keep its heading perfectly fixed on straight roads. It's a theoretical limitation of the algorithm. Unfortunately, I don't have a good idea of how to fix that :(.

closed 2021-08-10 by ke-sun