KumarRobotics/msckf_vio #121: world frame or inertial frame???

jingshaojing, 2021-08-10:
Sorry for my question again, I want to know what the **world frame** refers to in the code and the paper, 
in your paper ,_G_ frame is inertial frame, and in the code ,the corresponding state like Quaternion means a vector from the world frame to IMU frame.
So,  whether the world frame is an ECI frame or an ECEF frame, or an inertial frame, what is the conversion relationship between it and the ENU navigation frame?

Hope to get your answers  ! @ke-sun

ke-sun, 2021-08-10:
The roll and pitch of the world frame is defined so that the gravity vector in the world frame is aligned with the z-axis. The yaw of the world frame is 0. See the following lines for how the first IMU frame (which implicitly defines the world frame) is defined in the code:
https://github.com/KumarRobotics/msckf_vio/blob/68065ad48e6e6f411fdd64d153938a0130eb835f/src/msckf_vio.cpp#L278-L281

jingshaojing, 2021-08-11:
Next, I will describe my problem in detail, sorry that it may be lengthy and take up your time，my core question is how to transfer the posture from the world frame to the IMU frame,  to the ENU (East-North-Up) navigation system.
Please allow me to introduce my coordinate system. Below I will compare the posture at the first moment.  At the moment, my drone is placed stationary **towards the west** under the ENU frame, And I am equipped with a SPAN reference system, the output attitude of the reference is the attitude of the carrier system from the body frame(x-right,y-front,-up） to the ENU( The heading angle is _north to east is positive_)，the attitude is` [yaw, pitch, roll ] = [271.1291, 0.3046, 0.8173]`, You can see that this is correct.
The device I use to run S-MSCKF is D435i, and the installation position is also horizontally facing west (front view of the camera), and the aligned orientation of the benchmark system SPAN  , At this moment, the IMU in D435i is x toward right, y toward down, and z toward forward.
```
Quaterniond q0_i_w = Quaterniond::FromTwoVectors( 
   gravity_imu, -IMUState::gravity); 
 state_server.imu_state.orientation = 
   rotationToQuaternion(q0_i_w.toRotationMatrix().transpose()); 
```
From the above code, I calculated the attitude angle from IMU frame to word frame, which is  
`[yaw, pitch, roll ] = [2.893284146915304, 2.615595553602876,95.773536617404048]`
 I saved the pose from word frame to IMU frame as:  `[yaw, pitch, roll ] =[-1.14237340561959	-1.03271633907319	-95.7725871742850]
`, These two pose can verify each other that they are correct  .

So you can see that after the S-MSCKF runs , from IMU frame to world frame  that yaw is -1.14 degrees or so, but at this moment in the ENU coordinate system, the actual yaw should be about 271 degrees, so I should be how to align the two direction angles, or, from the word frame in codes to ENU ,how to convert  ??

This is a longer question, I'm sorry

closed 2021-09-04 by jingshaojing