KumarRobotics/msckf_vio #124: Chi-square table confidence value used for the gating test

zhangtao90, 2022-01-04:
Hi, I'm using some simulation generated data to run this vio back-end for more insights. Just found that the gating test seems working in a wrong status since the confidence value set to 0.05. In this setting almost all updates are rejected even if I set all noise params exactly as the simulated data. I just print out part of the chi-square table in this program as following:

chi-square table dof 1 tbl 0.00393214
chi-square table dof 2 tbl 0.102587
chi-square table dof 3 tbl 0.351846
chi-square table dof 4 tbl 0.710723
chi-square table dof 5 tbl 1.145483......

As in the code, you compute the NEES(normalized estimation error squared) as "r.transpose() * (HPH.transpose + R) * r" and compare it with the corresponding table value. It is obvious that, for example, you need a 1 dof measurement NEES less than 0.0039 to accept it as a valid update. It seems toooooo strict as you have an only 5% probability to accept a reasonable measurement ("reasonable" means the error or residual matches the covariance well so that NEES value follows the chi-square distribution). I further saw in the euroc launch file that the feature noise is tuned to 0.03(which is really big in intuition) to adapt to this gating rule.

I just change this value to 0.9 and tune the feature noise to a value close to the simulation ground truth and found obvious performance improvements when running with simulated data. I will further verify this with some datasets. Could you please help me verify your expected usage of chi-square table is just as same as mine or it is my misunderstanding?

ke-sun, 2022-01-06:
I think you are right. I intended to have 95% percentile for the chi square distribution in the table, but somehow I put 5%. With the 95% percentile (or 0.9 as you set), the feature noise parameter can be reduced to a more reasonable value. Actually I always wonder why the algorithm needs such a large feature noise (as you said). Thanks a lot for catching this.

zhangtao90, 2022-01-06:
OK, thanks for your verification !

closed 2022-02-15 by zhangtao90